tomeshnet/prototype-cjdns-pi #17: Mesh Interface Inactive after Installing with IPFS

garrying, 2016-09-07:
Ran the following:
`$ wget https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi2/master/scripts/install && chmod +x install && WITH_IPFS=true ./install`

benhylau, 2016-09-13:
I have not been able to reproduce this on a Pi 2 of 3. This seems like a rare occurrence.

nycoliver, 2016-09-13:
May be relevant: https://github.com/ipfs/ipfs/issues/189

claudiobizzotto, 2016-09-13:
Same thing here, @benhylau. Maybe we should just reference the issue pointed to by @nycoliver and close this one.

benhylau, 2016-09-14:
Wait. Is it `mesh0` or `tun0` that's inactive? I thought you meant `mesh0` and that's purely 802.11s mesh point with nothing to do with cjdns.

garrying, 2016-09-14:
It was `mesh0` being inactive. Closing this for now: I was running on a bleeding edge kernel/firmware and couldn't reproduce with stable.

closed 2016-09-14 by garrying