tomeshnet/prototype-cjdns-pi #20: WiFi adapter occasionally does not get brought up when Pi is turned on

benhylau, 2016-10-07:
This occurs infrequently, but has happened to a few people. Restarting the Pi 3 would fix it.

garrying, 2016-11-12:
I was able to reproduce on a fresh install of raspbian.

benhylau, 2016-11-18:
Nick mentioned something similar

benhylau, 2016-12-31:
I am finally able to reproduce this on the 2016-11-25 release of Raspbian Jessie Lite, SHA-1 `6587483c9b90b11185e2ce99175e27fe75af1f61`:

```
$ iw dev
phy#0
        Interface wlan1
                ifindex 4
                wdev 0x1
                addr f4:f2:6d:15:e1:a1
                type managed
phy#1
        Interface wlan0
                ifindex 3
                wdev 0x100000001
                addr b8:27:eb:dc:ac:a8
                type managed
```

On this particular Pi 3 device with this version of Raspbian, `phy#0` is the mesh point interface and is always mapped to `wlan1`, `phy#1` is the onboard and is always mapped to `wlan0`.

closed 2016-12-31 by benhylau