tomeshnet/prototype-cjdns-pi #33: Clean up after installing ipfs

uditvira, 2016-11-18:
The ipfs tarball and installation directory still exists after installation. This should be cleaned up after installation with a check to see if ipfs is working. 

`ipfs cat /ipfs/QmYwAPJzv5CZsnA625s3Xf2nemtYgPpHdWEz79ojWnPbdG/readme`

Prompt the user to see if the output looks like:

```
██║██╔══██╗██╔════╝██╔════╝
██║██████╔╝█████╗  ███████╗
██║██╔═══╝ ██╔══╝  ╚════██║
██║██║     ██║     ███████║
╚═╝╚═╝     ╚═╝     ╚══════╝

If you're seeing this, you have successfully installed
IPFS and are now interfacing with the ipfs merkledag!
```

benhylau, 2017-01-01:
The clean up is addressed by https://github.com/tomeshnet/prototype-cjdns-pi2/pull/44
As for status check, the `status` script currently checks `ipfs-daemon.service` status, perhaps `ipfs cat /ipfs/QmYwAPJzv5CZsnA625s3Xf2nemtYgPpHdWEz79ojWnPbdG/readme` can be added as part of the check?

benhylau, 2017-01-08:
Closing since cleanup is addressed and `ipfs-daemon.service` is checked

closed 2017-01-08 by benhylau