tomeshnet/prototype-cjdns-pi #35: Put systemd files in one consistent place

benhylau, 2016-11-18:
systemd `*.service` files are placed in both:

    /lib/systemd/system/
    /etc/systemd/system/

Put them in one consistent place.

darkdrgn2k, 2017-12-21:
> Basically, files that ships in packages downloaded from distribution repository go into /usr/lib/system/. Modifications done by system administrator (user) go into /etc/systemd/system/. System-specific units override units supplied by vendor. Using drop-ins, you can override only specific parts of unit file, leaving rest to vendor (drop-ins are available since the very beginning of systemd, but were properly documented only in v219; see man systemd.unit).

From what i read we should put them in /etc/systemd/system not /lib as these are not packages.

benhylau, 2017-12-21:
Consistent with what I have read earlier. Thanks!

closed 2017-12-21 by benhylau