tomeshnet/prototype-cjdns-pi #50: script for $9 single board computer CHIP

arkgit, 2017-01-24:
The C.H.I.P. would be a perfect fit for this script as well since it comes with wifi and is quite cheap than a pi2\3 

https://getchip.com

Famicoman, 2017-01-27:
This would likely be better in its own repo at something like prototype-cjdns-chip much like the odroidc2 project. I'd consider something similar for Beaglebones, etc.

benhylau, 2017-01-28:
Hoping to start a repo based on [dietpi](http://dietpi.com), with options to run an install script or build a custom image. We have already tried running cjdns on:

- Raspberry Pi Zero, 1, 2, 3
- ODROID XU4, C2
- Pine A64
- VirtualBox
- C.H.I.P. (not listed on dietpi support list at the moment, but I imagine it'd be fine)

So it seems a good idea to support all this hardware, and dietpi already got a lot of that covered. It also supports the [Orange Pi Zero](http://www.orangepi.org/orangepizero/). I am trying to get a hold of some at the moment, but it looks like exactly the right specs we need for a light-weight mesh node. Having the ethernet port with PoE, and one USB-A by default, with a USD 2 expansion to get two more USB-A, seem to be very useful for attaching all types of radios we need. We also suspect the on-board WiFi module supports 802.11s.

arkgit, 2017-01-28:
That sounds great. Once the install script for cjdns for the chip is ready
please point me to it

benhylau, 2017-03-13:
@arkgit we decided to build an image for the Orange Pi Zero instead https://github.com/tomeshnet/mesh-orange/issues/1 so no one is really working on something for the CHIP at the moment.

Famicoman, 2017-03-13:
@arkgit @benhylau I've had a CHIP back-ordered for a while now. When it comes I should be able to give everything a shot, and it probably makes the most sense to draw up a new repo somewhere instead of latching it onto an existing one.

benhylau, 2018-04-30:
Will not be supported

closed 2018-04-30 by benhylau