tomeshnet/prototype-cjdns-pi #54: ASUS USB-N14 seems to cause interface enumeration confusion (again)

benhylau, 2017-02-04:
**ASUS USB-N14** (used as Mesh Point interface) seems to make the access point not come up properly unless it is plugged in a few seconds after boot. Probably some variation of https://github.com/tomeshnet/prototype-cjdns-pi2/issues/20

**Setup:**
- Pi 3
- ASUS USB-N14
- Backhauled by a 3G connection

benhylau, 2017-02-04:
cc. @stigtsp

darkdrgn2k, 2017-09-04:
May be solved with modifications in #87 as it now has better phy to WLAN check

darkdrgn2k, 2017-09-17:
New mesh setup scripts no longer care about phy vs interface
#89 would solve the issue

darkdrgn2k, 2017-11-12:
to lock WLAN? location by mac

create a file /etc/udev/rules.d/10-network-device.rules

SUBSYSTEM=="net", ACTION=="add", ATTR{address}=="12:42:0d:80:70:50", NAME="wlan0"
SUBSYSTEM=="net", ACTION=="add", ATTR{address}=="8c:88:2b:00:00:0a", NAME="wlan1"
SUBSYSTEM=="net", ACTION=="add", ATTR{address}=="8c:88:2b:00:00:03", NAME="wlan2"

darkdrgn2k, 2018-04-16:
Even without rules.d this should not longer be a problem since we fixed the mesh script and we pin the Ap onboard adapter as wlan-ap by Mac address on boot.

Just need to confirm this is the case.. Still have the USB to test?

benhylau, 2018-04-16:
I don't think @stigtsp still has that USB. Let's close this and re-open if we encounter this again with other devices.

closed 2018-04-16 by benhylau