tomeshnet/prototype-cjdns-pi #60: Mesh interface?

arkgit, 2017-02-17:
Theres no instructions on readme about using the mesh interface

benhylau, 2017-02-17:
If you have two then they peer up automatically and you will see each other's cjdns IP on `status`

arkgit, 2017-02-17:
ah, thought there was more tools. /cjdns/tools has pingAll that is useful

arkgit, 2017-02-17:
The mesh interface says its not running when running status.

Also have you ran this over two nodes connected with directional antennas?

benhylau, 2017-02-17:
Do you have the 722N plugged in?

arkgit, 2017-02-17:
No, using two ubiquity antennas right now

benhylau, 2017-02-18:
Ahh, that's why the mesh interface says it's not running. You need devices that support 802.11s Mesh Point. README has more on this.

arkgit, 2017-02-18:
Ah ok. If i add the peers manually it should work though right?

benhylau, 2017-02-18:
Manually adding UDP peers, or having two devices connected to your router. Yes.

closed 2017-02-25 by benhylau