tomeshnet/prototype-cjdns-pi #61: ipfs installation not working

warningbagel, 2017-02-17:
when running the script with all the optional flags on or going to the directory and manually running the install script in prototype-cjdns-pi2/scripts/ipfs i get
>./install: line 18: /usr/local/bin/ipfs: cannot execute binary file: Exec format error

benhylau, 2017-02-18:
Pi 3?

JoshuaOpolko, 2017-02-18:
The error might indicate that the ipfs binary wasn't compiled for the right architecture. 
You can confirm this by typing 'file /usr/local/bin/ipfs' and see if the output matches that below:
"ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, not stripped"

Otherwise, I've tested the install successfully with a pristine raspian lite on a pi2. The wget script with all options set to true completed successfully and mesh, cjdns and ipfs are all active after reboot with no intervention from me. I'd suggest reinstalling the os and re-launching the install script in case any previous modifications/flags/env variables are affecting the script.

For further issues we can trace the install and see where it fails. Some of this will depend on whether the ipfs binary is actually being installed at /usr/local/bin or not.

benhylau, 2017-02-19:
Thanks for verifying @JoshuaOpolko

closed 2017-02-25 by benhylau