tomeshnet/prototype-cjdns-pi #66: package installation failed.

Wtango, 2017-03-05:
During the installation i got an error:

> sed: -e expression #1, char 1: unknown command: `,'

Maybe this error comes from a third part software package, but i don't know which package it is.

The trace log show as bellow:

> Build completed successfully, type ./cjdroute to begin setup.
Total build time: 176112ms.
Created symlink from /etc/systemd/system/multi-user.target.wants/cjdns.service to /lib/systemd/system/cjdns.service.
Created symlink from /etc/systemd/system/sleep.target.wants/cjdns-resume.service to /lib/systemd/system/cjdns-resume.service.
Created symlink from /etc/systemd/system/multi-user.target.wants/mesh.service to /lib/systemd/system/mesh.service.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  apt-clone archdetect-deb dpkg-repack gir1.2-json-1.0 gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 libdebian-installer4 libparted-fs-resize0
  libtimezonemap-data libtimezonemap1 os-prober python3-icu python3-pam rdate
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libnl-route-3-200
The following NEW packages will be installed:
  dnsmasq hostapd libnl-route-3-200 radvd
0 upgraded, 4 newly installed, 0 to remove and 345 not upgraded.
Need to get 615 kB of archives.
After this operation, 1,531 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com xenial-updates/universe armhf dnsmasq all 2.75-1ubuntu0.16.04.1 [15.9 kB]
Get:2 http://ports.ubuntu.com xenial/main armhf libnl-route-3-200 armhf 3.2.27-1 [104 kB]
Get:3 http://ports.ubuntu.com xenial/universe armhf hostapd armhf 1:2.4-0ubuntu6 [439 kB]                                                     
Get:4 http://ports.ubuntu.com xenial/main armhf radvd armhf 1:2.11-1 [56.0 kB]                                                                
Fetched 615 kB in 47s (13.0 kB/s)                                                                                                             
Selecting previously unselected package dnsmasq.
(Reading database ... 175211 files and directories currently installed.)
Preparing to unpack .../dnsmasq_2.75-1ubuntu0.16.04.1_all.deb ...
Unpacking dnsmasq (2.75-1ubuntu0.16.04.1) ...
Selecting previously unselected package libnl-route-3-200:armhf.
Preparing to unpack .../libnl-route-3-200_3.2.27-1_armhf.deb ...
Unpacking libnl-route-3-200:armhf (3.2.27-1) ...
Selecting previously unselected package hostapd.
Preparing to unpack .../hostapd_1%3a2.4-0ubuntu6_armhf.deb ...
Unpacking hostapd (1:2.4-0ubuntu6) ...
Selecting previously unselected package radvd.
Preparing to unpack .../radvd_1%3a2.11-1_armhf.deb ...
Unpacking radvd (1:2.11-1) ...
Processing triggers for systemd (229-4ubuntu10) ...
Processing triggers for ureadahead (0.100.0-19) ...
Processing triggers for libc-bin (2.23-0ubuntu3) ...
Processing triggers for man-db (2.7.5-1) ...
Setting up dnsmasq (2.75-1ubuntu0.16.04.1) ...
Setting up libnl-route-3-200:armhf (3.2.27-1) ...
Setting up hostapd (1:2.4-0ubuntu6) ...
Setting up radvd (1:2.11-1) ...
dpkg-statoverride: warning: --update given but /var/run/radvd does not exist
Processing triggers for systemd (229-4ubuntu10) ...
Processing triggers for ureadahead (0.100.0-19) ...
Processing triggers for libc-bin (2.23-0ubuntu3) ...
sed: -e expression #1, char 1: unknown command: `,'
root@my-desktop:~#

benhylau, 2017-03-05:
This repo is for Raspbian. See README

Wtango, 2017-03-06:
Can you point out where this error comes from? I try to fix it in Ubuntu.

Wtango, 2017-03-06:
Hi, I have a look into this problem. Maybe this error comes from [hostapd install](https://github.com/tomeshnet/prototype-cjdns-pi2/blob/master/scripts/hostapd/install#L16).
Since I configure wlan0 interface by **/etc/network/interfaces.d/wlan0**  , so the variable $START get an empty string, and _sed_ complain this error.

benhylau, 2017-03-06:
Yes, all the commands assume Raspbian, and I try to make sure this works with the latest Raspbian release as they come out. I wouldn't be surprised if more things fail running this on Ubuntu. If you really need to use Ubuntu, perhaps it's more appropriate to do that in a fork rather than switching commands based on Debian or Ubuntu?

closed 2017-03-08 by benhylau