tomeshnet/prototype-cjdns-pi #85: Test Raspbian Stretch

benhylau, 2017-09-01:
https://www.raspberrypi.org/downloads/raspbian/

darkdrgn2k, 2017-09-05:
Ran the latest DEVELOP branch install on a clean RPI 1 B+ using stretch and it worked without a hitch!

I'm not sure if this is the same in previous releases but the boot is on a separate partition. 
sdx1 was boot partition
sdx2 was root partition

So
$ touch /path/to/sd/boot/ssh 
became
$ touch /path/to/sdx1/ssh

darkdrgn2k, 2017-09-05:
Few things in running it

Predictable Interface Naming is a problem just like on the orange pis, but thats an easy fix

The bigger problem seems to be they left out mesh point support!

```
+ sudo iw dev mesh0 mesh join tomesh freq 2412 HT40+
command failed: Operation not supported (-95)
```

benhylau, 2017-09-05:
Even with setting the region it fails?
https://github.com/tomeshnet/prototype-cjdns-pi/blob/master/scripts/mesh-point/mesh#L6

darkdrgn2k, 2017-09-05:
I pushed a change to the install2 script that will correct that problem.

it was the PREDICTABLE INTERFACE NAMING that i encountered in OPI where it names the usb interfaces a wierd name.

pull the git and copy the mesh file over to /usr/bin  should be fine then

darkdrgn2k, 2017-09-05:
To avoid a reinstall just run this and reboot and the mesh script will work again (less meshpoint)

`sudo touch /etc/udev/rules.d/80-net-setup-link.rules`

darkdrgn2k, 2017-09-05:
Regarding MeshPoint

I have posted in Raspbian forum waiting for response
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=192531&p=1207093#p1207093

When I  encountered with with OPIZ H5 it was a missing kernel option
`CONFIG_MAC80211_MESH`

Looking at the git hub
https://github.com/raspberrypi/linux
It SEEMS like some arch have it enabled others done but i'm not quite sure how this kernel gets implemented with RASPBIAN.

darkdrgn2k, 2017-09-22:
SO i found out that mesh is in fact enabled in stretch

Done know why im getting an error trying to bring it up!

to check configs on a pi:
```
$ sudo modprobe configs
$ sudo zcat /proc/config.gz  | grep -i mesh
CONFIG_MAC80211_MESH=y
```

darkdrgn2k, 2017-09-26:
Everything seems to work on stretch so far :)

benhylau, 2017-10-19:
On Raspberry Pi 3, we need `NO_TEST` for cjdns to compile. Keep this issue open until that is sorted.
https://github.com/tomeshnet/prototype-cjdns-pi/commit/3fd9c0c4b0fba49b0ad1b7aaa87623b4f3bf7b01

benhylau, 2017-10-20:
Actually, only `--bench` is working. When you actually start `cjdroute` it crashes:

```
pi@raspberrypi:/opt/cjdns $ sudo ./cjdroute < /etc/cjdroute.conf
1508481072 INFO cjdroute2.c:642 Cjdns armel linux +seccomp
1508481072 INFO cjdroute2.c:646 Checking for running instance...
1508481072 DEBUG UDPAddrIface.c:293 Bound to address [0.0.0.0:55003]
1508481072 DEBUG AdminClient.c:333 Connecting to [127.0.0.1:11234]
1508481072 DEBUG Pipe.c:134 Buffering a message
1508481072 DEBUG cjdroute2.c:699 Sent [140] bytes to core
1508481072 INFO RandomSeed.c:42 Attempting to seed random number generator
1508481072 INFO RandomSeed.c:50 Trying random seed [/dev/urandom] Success
1508481072 INFO RandomSeed.c:50 Trying random seed [sysctl(RANDOM_UUID) (Linux)] Success
1508481072 INFO RandomSeed.c:50 Trying random seed [/proc/sys/kernel/random/uuid (Linux)] Success
1508481072 INFO RandomSeed.c:50 Trying random seed [getentropy(2)] Success
1508481072 INFO RandomSeed.c:64 Seeding random number generator succeeded with [4] sources
1508481072 INFO LibuvEntropyProvider.c:59 Taking clock samples every [1000]ms for random generator
1508481072 DEBUG Core.c:354 Getting pre-configuration from client
1508481072 DEBUG Pipe.c:231 Pipe [/tmp/cjdns_pipe_client-core-b1yum4st7mz67dlu8s2046j372bt8n] established connection
1508481072 DEBUG Pipe.c:231 Pipe [/tmp/cjdns_pipe_client-core-b1yum4st7mz67dlu8s2046j372bt8n] established connection
1508481072 DEBUG Pipe.c:253 Sending buffered message
1508481072 DEBUG Core.c:357 Finished getting pre-configuration from client
1508481072 DEBUG UDPAddrIface.c:254 Binding to address [127.0.0.1:11234]
1508481072 DEBUG UDPAddrIface.c:293 Bound to address [127.0.0.1:11234]
1508481072 DEBUG UDPAddrIface.c:293 Bound to address [0.0.0.0:44551]
1508481072 DEBUG AdminClient.c:333 Connecting to [127.0.0.1:11234]

Core exited with status [0], signal [7]
Aborted
```

benhylau, 2017-10-20:
Mesh interface is also broken on Stretch :(

cjdelisle, 2017-10-20:
Can you please GDB this, it's an unaligned memory access, also please check with crashey.

darkdrgn2k, 2017-10-20:
" Mesh interface is also broken on Stretch + Raspberry Pi 3 :("

What broken on it? Seems to work for me!

root@raspberrypi:/opt# iw dev mesh0 info
Interface mesh0
        ifindex 5
        wdev 0x2
        addr 18:d6:c7:0d:78:94
        type managed
        wiphy 0
        txpower 30.00 dBm

darkdrgn2k, 2017-10-20:
To test if MESH is supported on your install run the following

` sudo modprobe configs
`
`zcat  /proc/config.gz  | grep MESH
`

you should see

`CONFIG_MAC80211_MESH=y
`

darkdrgn2k, 2017-10-20:
root@raspberrypi:/usr/bin# iw --debug dev mesh0 mesh join tomesh freq 2412 
-- Debug: Sent Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 48
    .type = 22 <0x16>
    .flags = 5 <REQUEST,ACK>
    .seq = 1508506231
    .port = 1497367426
  [GENERIC NETLINK HEADER] 4 octets
    .cmd = 68
    .version = 0
    .unused = 0
  [PAYLOAD] 28 octets
    08 00 03 00 05 00 00 00 0a 00 18 00 74 6f 6d 65 ............tome
    73 68 00 00 08 00 26 00 6c 09 00 00             sh....&.l...
---------------------------  END NETLINK MESSAGE   ---------------------------
-- Debug: Received Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 68
    .type = 2 <ERROR>
    .flags = 0 <>
    .seq = 1508506231
    .port = 1497367426
  [ERRORMSG] 20 octets
    .error = -95 "Operation not supported"
  [ORIGINAL MESSAGE] 16 octets
    .nlmsg_len = 16
    .type = 22 <0x16>
    .flags = 5 <REQUEST,ACK>
    .seq = 1508506231
    .port = 1497367426
---------------------------  END NETLINK MESSAGE   ---------------------------
command failed: Operation not supported (-95)

benhylau, 2017-10-29:
All issues should be addressed on current develop. @darkdrgn2k please close this Issue if this sounds right to you too.

closed 2017-10-30 by darkdrgn2k