tomeshnet/prototype-cjdns-pi #112: IPFS on Orange Pi Zero

benhylau, 2017-11-02:
Probably applies to other Orange Pi boards on Armbian too. Running the installation as `root` user, IPFS service doesn't start. Suspicion is something wrong here:

https://github.com/tomeshnet/prototype-cjdns-pi/blob/develop/scripts/ipfs/install#L18
https://github.com/tomeshnet/prototype-cjdns-pi/blob/develop/scripts/ipfs/ipfs.service#L8

Current workaround is to manually run this after install and restart the service:

```
rm -rf /root/.ipfs
ipfs init
```

benhylau, 2017-11-30:
@claudiobizzotto This is the issue I was talking about

claudiobizzotto, 2017-12-02:
@benhylau :+1: I'll keep an eye on this one.

claudiobizzotto, 2017-12-02:
Finally got the [installer](https://github.com/claudiobizzotto/ipfs-rpi/tree/v0.1.0) working again. I'm now using the pre-built ARM binary instead of building everything from source.
Anyways, I'm going to test that on the Orange Pi some time this weekend.
I may also add Nginx to that installer as a separate pair of scripts (installer + uninstaller). What do you think of using all that here?

benhylau, 2017-12-02:
You mean nginx for the HTTP gateway?
https://github.com/tomeshnet/prototype-cjdns-pi/blob/master/scripts/ipfs/ipfs-http-gateway.nginx

Porting that to a separate install script in https://github.com/claudiobizzotto/ipfs-rpi/ right? Is the gateway reachable on your LAN without nginx?

claudiobizzotto, 2017-12-03:
Not by default. By default, this will work (from local machine, i.e. RPi):
```SHELL
$ curl -i http://localhost:8080/ipfs/QmYwAPJzv5CZsnA625s3Xf2nemtYgPpHdWEz79ojWnPbdG/readme
```

but his won't (from another machine in LAN, i.e. my laptop):
```SHELL
$ curl -i http://xxx.xxx.x.x:8080/ipfs/QmYwAPJzv5CZsnA625s3Xf2nemtYgPpHdWEz79ojWnPbdG/readme
```

*But*, for the second one to work, all we need to do is tell IPS to accept requests from all network interfaces on its API and web interface (`~/.ipfs/config`):
```
"API": "/ip4/0.0.0.0/tcp/5001",
"Gateway": "/ip4/0.0.0.0/tcp/8080"
```

IPFS comes with a web server listening on those ports by default.

Are you guys using Nginx just for web/API access? If that's the case, you could get rid of it if you wanted.

benhylau, 2017-12-03:
Yes I was using nginx just for that. The first part is consistent with my experience, and I didn't know about the `~/.ipfs/config` before. Can I list both ip4 and ip6, I need it reachable over the cjdns IP.

claudiobizzotto, 2017-12-03:
It looks like we can change both the port and the protocol (eg.: `"Gateway": "/ip6/::/tcp/8080"`), but I'm afraid there's no way to specify both IP4 and IP6 at the same time. I'll double check.

benhylau, 2017-12-03:
We can probably live with ip6-only for this repo, but may as well keep nginx in for the flexibility.

claudiobizzotto, 2017-12-21:
We might keep Nginx, after all, @benhylau. I can now confirm that:
* IPFS gateway (web UI) works OK with IPv6
* you _can't_ have it serve on both IPv4 and IPv6

As for the IPFS-only stuff, I released [fixes](https://github.com/claudiobizzotto/ipfs-rpi/releases/tag/v0.1.1) that make the installer work with `upstart` as well. Also, it's all running beautifully on the Orange Pi (Ubuntu 14.04).

How do you think we should use that here? Git submodule, Git subtree?

benhylau, 2017-12-21:
Either we `git clone` from https://github.com/claudiobizzotto/ipfs-rpi or edit the scripts here directly. Armbian's Ubuntu uses Upstart?

@darkdrgn2k thoughts?

benhylau, 2018-04-16:
Yes that's what I do for temporary things I want to serve over cjdns, but feel nginx has better flexibility for persistent services.

darkdrgn2k, 2018-08-16:
Seems to be working on orange pi zero.

closed 2018-08-21 by benhylau