tomeshnet/prototype-cjdns-pi #116: Toplinkst TOP-GS07/rt2800usb Issues

darkdrgn2k, 2017-11-10:
Issues/Possible Solutions documented for the rt2800usb driver and the Toplinkst

darkdrgn2k, 2017-11-10:
Device increases the CPU heat of the orange pi board by about 10-15 degrease, board itself gets very hot. We also see the board cut out seemingly when it gets to hot.

On the orange pi we traced some of the cutouts to a periodic scan (see #113)

darkdrgn2k, 2017-11-10:
Adding a fan to blow on the board reduced heating significantly

Found bard still drops throughput showing dmesg errors

```
[  539.964921] ieee80211 phy1: rt2800usb_txdone: Warning - Data pending for entry 10 in queue 2
[  539.968312] ieee80211 phy1: rt2800usb_txdone: Warning - Data pending for entry 12 in queue 2
[  545.275751] ieee80211 phy1: rt2800usb_txdone: Warning - Data pending for entry 14 in queue 2
[  545.276666] ieee80211 phy1: rt2800usb_txdone: Warning - Data pending for entry 15 in queue 2
[  552.796993] ieee80211 phy1: rt2800usb_txdone: Warning - Data pending for entry 11 in queue 2
[  557.632021] ieee80211 phy1: rt2800usb_txdone: Warning - Data pending for entry 1 in queue 2
[  557.632241] ieee80211 phy1: rt2800usb_txdone: Warning - Data pending for entry 1 in queue 2
```

Seems to occur without any reason or ryme

darkdrgn2k, 2017-11-10:
tried the follownig and it seems to have stabilised (not sure which part did it) Board holing a steady on average of 20-30 MBps with single second drops to single digits here and there.

- Turned off power management
- turned off hardware crypto using options rt2800usb nohwcrypt=1
- Found following settings for board

```
iwconfig wlan1 retry 30
iwconfig wlan1 rts 2347
iwconfig wlan1 frag 2346
```

benhylau, 2017-11-17:
With two nodes < 1 m apart, with fan cooling on each GS07, what is the speed range you get and is it stable?

benhylau, 2017-11-17:
Without cooling at 5 GHz you get drops like this. This usually happens after 30 s of transmission and stays at ~2 Mbps until adapter is cooled.

```
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  4.68 MBytes  39.2 Mbits/sec   37   45.7 KBytes
[  4]   1.00-2.00   sec  2.69 MBytes  22.6 Mbits/sec    0   75.8 KBytes
[  4]   2.00-3.00   sec  1001 KBytes  8.20 Mbits/sec   38   21.7 KBytes
[  4]   3.00-4.00   sec   813 KBytes  6.66 Mbits/sec   26   31.3 KBytes
[  4]   4.00-5.00   sec   438 KBytes  3.59 Mbits/sec   14   24.1 KBytes
[  4]   5.00-6.00   sec   876 KBytes  7.17 Mbits/sec   26   22.9 KBytes
[  4]   6.00-7.00   sec   375 KBytes  3.07 Mbits/sec   20   18.0 KBytes
[  4]   7.00-8.00   sec   313 KBytes  2.56 Mbits/sec   26   19.2 KBytes
[  4]   8.00-9.00   sec   500 KBytes  4.10 Mbits/sec   28   10.8 KBytes
[  4]   9.00-10.00  sec   500 KBytes  4.10 Mbits/sec   32   8.42 KBytes
[  4]  10.00-11.00  sec   375 KBytes  3.08 Mbits/sec   31   13.2 KBytes
[  4]  11.00-12.00  sec   375 KBytes  3.07 Mbits/sec   31   9.62 KBytes
[  4]  12.00-13.00  sec   375 KBytes  3.08 Mbits/sec   12   9.62 KBytes
[  4]  13.00-14.00  sec   438 KBytes  3.59 Mbits/sec   22   13.2 KBytes
[  4]  14.00-15.00  sec   313 KBytes  2.56 Mbits/sec   32   10.8 KBytes
[  4]  15.00-16.00  sec   313 KBytes  2.56 Mbits/sec   24   12.0 KBytes
[  4]  16.00-17.00  sec   250 KBytes  2.05 Mbits/sec   26   10.8 KBytes
[  4]  17.00-18.00  sec   313 KBytes  2.56 Mbits/sec   19   13.2 KBytes
[  4]  18.00-19.00  sec   250 KBytes  2.05 Mbits/sec   31   10.8 KBytes
[  4]  19.00-20.00  sec   250 KBytes  2.05 Mbits/sec   19   13.2 KBytes
```

darkdrgn2k, 2018-08-16:
Toplinkst-top-s5 seems to work a bit better.  However both dongles are now end of life.

benhylau, 2018-09-30:
We are still able to buy TOP-S5. I think that is the solution to this.

closed 2018-09-30 by benhylau