tomeshnet/prototype-cjdns-pi #117: Not getting through.

paappraiser, 2017-11-20:
Fresh install stretch. 

All nodes and peers communicate to each other perfectly. Connect to main node connected to LAN(internet) internet is connected and fine. Connect to any other peer and no internet. 

Been running into problems with stretch. 

Do I have to turn off "WITH_WIFI_AP" on the peers? 

Sorry if noob question, wifi is my nemesis.

benhylau, 2017-11-20:
Yes the other peers are only on the mesh net. You can reach them only via another cjdns node (e.g. your main node). There is a feature in cjdns called iptunnel, which allows for your mesh-only node to access internet through another internet-connected node that functions as a gateway. We will soon incorporate that into the prototype script, but @darkdrgn2k is currently running a gateway, maybe he can give details on how to hop on if you want to test.

`WITH_WIFI_AP` is unrelated to this. It just uses your Raspberry Pi 3's onboard WiFi to create an access point, so your other devices will see a regular `tomesh-xxxx` access point if you have that turned on. Any device connected to that access point will have internet if the node itself has internet, otherwise it will only have access to the in-mesh resources at `fc00::/8` IP addresses.

Feel free to hop on to https://chat.tomesh.net for support questions.

darkdrgn2k, 2017-11-20:
As @benhylau mentioned there is "INTERNET" on CJDNS. CJDNS has its own "INTERNET" called hyperborea that is currently mostly peered over the internet ( see http://fc00.org ) You need to find some one to peer with thats already connected.

There is the "illusion" of internet by the use of the ACCESS POINT but all that does it provide you with internet through your existing internet bypassing CJDNS altogether. This illusion breaks the second you unplug your internet from the node.

To actually get route your internet to your node over CJDNS you have to use IPTUNNEL that will "tunnel" the internet over CJDNS to your node. This requires a SERVER/CLIENT setup similar to a VPN.

This is done as usually there is a COST associated with the internet and you don't want just anyone using your pipe!

Server Side (node that HAS internet)

in your cjdroute.conf
under "ipTunnel", "AllowedConnection" add the NODE information. 
This will require your PUBLIC KEY you can get from the "clients" cjdroute.conf file
IP Address is a IPv4 address you will assign to the CLIENT node
you need to do this for EVERY node you want intrent on
Example
```
"allowedConnections":
[
    {
                     "publicKey": "3wc3p4dsx6t2ykvp6xwsrqghxpd1sk925llvkvhtvnljdpd6zw30.k",
                     "ip4Address": "192.168.1.25"
                     "ip4Prefix": 0,
     },

]
```

Next restart your cjdroute 
if your using our script just do killall cjdroute 
it will automatically restart

once restarted you need to config your server with a NAT, ROUTE and IP ADDRESS
example assumes your ipv4 addresses above are 192.168.1.x and your internet is on eth0
```
ifconfig tun0 192.168.1.1
route add -net 192.168.1.0/24 tun0
iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
```

**NOTE** you must run the above each time cjdroute starts so you may want to add it to your /etc/rc.local

CLIENT SIDE

again in cjdroute
under "ipTunnel", "outgoingConnections"  add the PUBLIC KEY of your SERVER like so

```
  "outgoingConnections":
            [
                // Connect to one or more machines and ask them for IP addresses.
                 "g91lxyxhq0kExampleExampleExampleE5ample630mknuhw75l0.k"
            ]

```

then restart your cjdroute (killall cjdroute) and if you do an ifconfig you should see tun0 get the ip address assigned to it in the server

And you now have interent access on your node

darkdrgn2k, 2017-11-20:
So that was the DIY solution.

If your already peered to hyperborea and want quick internet access come by let us know ill show you how to connect to my test node :)

benhylau, 2017-11-20:
@paappraiser For example, you can have the "main" node with internet act as a server. Then the other nodes can be iptunnel clients, by pinning the gateway server address as @darkdrgn2k described.

Now if you have `WITH_WIFI_AP` enabled on the client, and you want your phone connected to that access point to have internet, you'd need to run these as well on the client:

```
sudo iptables -t nat -D POSTROUTING -o eth0 -j MASQUERADE
sudo iptables -t nat -A POSTROUTING -o tun0 -j MASQUERADE
```
That tells the nat to stop routing through `eth0` and route through the cjdns `tun0`, which cjdns then routes to the main node, and because of iptunnel, it will fetch internet traffic for you over its `eth0`.

paappraiser, 2017-11-20:
Excellent write up!..  ill keep you updated.

darkdrgn2k, 2018-01-04:
@paappraiser any update on this issue?

paappraiser, 2018-01-04:
Hi- Sorry for the lack of update. I ordered a bunch of compatible wifi modules from China and they just came in the other week. I will be working on this again

darkdrgn2k, 2018-01-04:
Great!  If they are [dongles](https://github.com/phillymesh/802.11s-adapters) we haven't tested let us know how it goes.  Or even feed back in general!

Also you can visit us in our  [chat room](https://chat.tomesh.net/#/room/#tomesh:tomesh.net) its usually quite active.

darkdrgn2k, 2018-04-15:
Perhaps the writeup can be moved to some sort of general docs folder and the issue closed?

darkdrgn2k, 2018-08-16:
Question answered.
Closing because of inactivity. 
Re-Open if needed

closed 2018-08-16 by darkdrgn2k