tomeshnet/prototype-cjdns-pi #119: Installation fails after Grafana is installed

TwistTheNeil, 2017-12-15:
Hi, I tried installing on a fresh raspbian install and got an error during the install. I'm not sure if it's related to grafana but it failed connecting to localhost:3000

```
.
.
.
electing previously unselected package grafana.
(Reading database ... 38934 files and directories currently installed.)
Preparing to unpack .../grafana/tmp/go-grafana.tar.gz ...
Unpacking grafana (4.3.2) ...
Setting up grafana (4.3.2) ...
Adding system user `grafana' (UID 111) ...
Adding new user `grafana' (UID 111) with group `grafana' ...
Not creating home directory `/usr/share/grafana'.
### NOT starting on installation, please execute the following statements to configure grafana to start automatically using systemd
 sudo /bin/systemctl daemon-reload
 sudo /bin/systemctl enable grafana-server
### You can start grafana-server by executing
 sudo /bin/systemctl start grafana-server
Processing triggers for systemd (232-25+deb9u1) ...
Synchronizing state of grafana-server.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable grafana-server
Created symlink /etc/systemd/system/multi-user.target.wants/grafana-server.service → /usr/lib/systemd/system/grafana-server.service.
curl: (7) Failed to connect to localhost port 3000: Connection refused
```

Related syslog:
```
.
.
.
Dec 15 00:50:12 rpi3-1 grafana-server[9923]: t=2017-12-15T00:50:12-0500 lvl=info msg="Executing migration" logger=migrator id="create test_data table"
Dec 15 00:50:15 rpi3-1 grafana-server[9923]: t=2017-12-15T00:50:15-0500 lvl=info msg="Created default admin user: admin"
Dec 15 00:50:15 rpi3-1 grafana-server[9923]: t=2017-12-15T00:50:15-0500 lvl=info msg="Starting plugin search" logger=plugins
Dec 15 00:50:15 rpi3-1 grafana-server[9923]: t=2017-12-15T00:50:15-0500 lvl=warn msg="Plugin dir does not exist" logger=plugins dir=/var/lib/grafana/plugins
Dec 15 00:50:15 rpi3-1 grafana-server[9923]: t=2017-12-15T00:50:15-0500 lvl=info msg="Plugin dir created" logger=plugins dir=/var/lib/grafana/plugins
Dec 15 00:50:15 rpi3-1 grafana-server[9923]: t=2017-12-15T00:50:15-0500 lvl=info msg="Initializing Alerting" logger=alerting.engine
Dec 15 00:50:15 rpi3-1 grafana-server[9923]: t=2017-12-15T00:50:15-0500 lvl=info msg="Initializing CleanUpService" logger=cleanup
Dec 15 00:50:15 rpi3-1 grafana-server[9923]: t=2017-12-15T00:50:15-0500 lvl=info msg="Initializing Stream Manager"
Dec 15 00:50:15 rpi3-1 grafana-server[9923]: t=2017-12-15T00:50:15-0500 lvl=info msg="Initializing HTTP Server" logger=http.server address=0.0.0.0:3000 protocol=http subUrl= socket=
```

systemctl status grafana-server:
```
pi@rpi3-1:~ $ sudo systemctl status grafana-server
● grafana-server.service - Grafana instance
   Loaded: loaded (/usr/lib/systemd/system/grafana-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2017-12-15 00:49:35 EST; 1h 35min ago
     Docs: http://docs.grafana.org
 Main PID: 9923 (grafana-server)
   CGroup: /system.slice/grafana-server.service
           └─9923 /usr/sbin/grafana-server --config=/etc/grafana/grafana.ini --pidfile= cfg:default.paths.logs=/var/log/grafana cfg:default.paths.data=/var/lib/grafana cfg:default.paths.plugins=/var/lib/grafana/plugins
```

darkdrgn2k, 2017-12-15:
Very strange.
It looks like the grafana server did not start to be able to run API calls against it.

Can you confirm which version of raspbian you did (Lite of other)
Also if you installed DEVEL or MASTER

Also can you see if you have a home directory for grafana, it didn't create one im wondering why

```Not creating home directory `/usr/share/grafana'.```

TwistTheNeil, 2017-12-15:
> Can you confirm which version of raspbian you did (Lite of other)

I used raspbian lite

> Also if you installed DEVEL or MASTER

From master. I just copy pasted the wget && install line from the tomesh website

> Also can you see if you have a home directory for grafana, it didn't create one im wondering why

It exists
```
pi@rpi3-1:~ $ cd /usr/share/grafana/
pi@rpi3-1:/usr/share/grafana $ ls
conf  public  scripts  vendor  VERSION
```

darkdrgn2k, 2017-12-15:
So it looks like grafana just took longer then normal to start (a fluke)

Perhaps we need to do some smarter test then just waiting 30 seconds

Ok so you can re-image and re-install and it SHOULD 

or finish the install of grafana manually
(id run these commands individually because im doing most of this from memory:))
```
cd prototype-cjdns-pi
cd scripts
cd grafana
curl --user admin:admin -X POST -H 'Content-Type: application/json' --data-binary @datasource.json http://localhost:3000/api/datasources
curl --user admin:admin -X POST -H 'Content-Type: application/json' --data-binary @dashboard.json http://localhost:3000/api/dashboards/db
cd ..
cd ..
```

Then re-start the script
```
cd  prototype-cjdns-pi
cd scripts
./install2
```

Say NO to everything up to and including grafana
then yes to whatever else you want

./install2
```

TwistTheNeil, 2017-12-17:
I thought that the 30 second wait would be it - I'll give your suggestion a shot in the morning.

Just out of curiosity, instead of sleeping 30, would it be possible to check the status of grafana before attempting to run the API calls to it?

darkdrgn2k, 2017-12-17:
Yes you are correct we should make it more robust.  

This is one of the issues with using scripts against a fully loaded os.

I will look into how we can correct the problem in the future if all else works for you . Thank you for the report.

benhylau, 2018-04-16:
@darkdrgn2k Please close this if you think it's sufficiently addressed by 38ea838, otherwise if you have a longer term fix in mind we can keep this open.

closed 2018-04-16 by TwistTheNeil

benhylau, 2018-04-16:
Thanks @TwistTheNeil