tomeshnet/prototype-cjdns-pi #120: Ornage pi - Log

darkdrgn2k, 2017-12-15:
After running for a long time the ramfs log directory gets full
```
log2ram          50M   50M     0 100% /var/log
```

benhylau, 2017-12-17:
Remove logs periodically?

darkdrgn2k, 2017-12-17:
Yeh, 
most of them the files there where  gz and didnt take up allof of space bug there where the .1 that ate up most of it.

darkdrgn2k, 2018-04-16:
Most of the logs when I noticed  where 
XXXX.1.gz 
XXXX.2.gz
etc

The biggest files however  eating up the most space where not gziped
XXXX and XXXX.1

perhaps we just need a cleanup script that delete *.1 and *.gz from lgo.

darkdrgn2k, 2019-02-12:
Long running orange pi nodes dont seem to have this problem anymore.

darkdrgn2k, 2019-02-13:
Long running orange pi 100% space

Fix:
`rm -rf /var/log/*.?.*`

makeworld-the-better-one, 2019-02-13:
I have had the same issue with my Raspberry Pi, and @dasanchez too I believe. Maybe we should include this in out stack somehow?