tomeshnet/prototype-cjdns-pi #128: prometheus-node-exporter Rename

darkdrgn2k, 2017-12-21:
prometheus-node-exporter is long (esp in the readme)
the program is just called node-exporter

perhaps drop the prometheus part, we don't do prometheus-grafana 

thoughts?

benhylau, 2017-12-21:
But node-exporter is part of Prometheus, from what I understand. Grafana is a separate project.

darkdrgn2k, 2017-12-22:
Yeh its more of a cosmetic thing to not have a 50 letter flag in the readmes and such.

Close the issue if you like or tell me and ill make the necessary changes.

Really I'm impartial. Your Call:)

benhylau, 2017-12-22:
To not deviate from the convention (since prometheus-server has the prometheus in there), and node-exporter being too generic of a name, I think we should keep as is.

closed 2017-12-22 by benhylau