tomeshnet/prototype-cjdns-pi #142: Update documentation for Pi 3 B+ support

skylarmt, 2018-03-23:
Is the Pi 3 B+ supported?

darkdrgn2k, 2018-04-02:
Sorry for the delay in responding to this issue. Yes i have tested it and it works. It is also slightly faster then the pi3 B+. 

I have not had time to really test the board so please let us know your results.

I'd suggest using [develop-bata](https://github.com/tomeshnet/prototype-cjdns-pi/tree/develop-beta) branch. I have already added detection for PI 3b+ into it

```
$ wget https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi/develop-beta/scripts/install && chmod +x install && TAG_PROTOTYPE_CJDNS_PI=develop-beta ./install
```

darkdrgn2k, 2018-04-15:
#141 merged into the Develop now  and slated for Version 2

@skylarmt any more issues or can we close.

skylarmt, 2018-04-15:
Nothing else, the code itself worked fine when I tried it before your commits, the issue was mostly about documentation.

closed 2018-04-15 by skylarmt