tomeshnet/prototype-cjdns-pi #151: Raspberry pi 3b plus Access point won't nat cjdns

darkdrgn2k, 2018-05-02:
Issue: Using Raspberry Pi3B+'s Access point to access a different FC00/8 node

I confined that packets are coming in, and hit the forward chain correctly and look like they are forwarded across the right interface

`May  2 03:18:31 tomesh-8668 kernel: [ 2749.468856] IN=wlan-ap OUT=tun0 MAC=b8:27:eb:13:76:3a:44:2c:05:df:98:92:86:dd SRC=fdfc:0000:0000:0000:dd16:d9ce:18f1:341b DST=fcaa:5785:a537:90db:6513:bba9:87a0:12a7 LEN=88 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=257`

However the packets don't seem to come through the POSTROUTING chain at all.

However I do see some activity for NONE fc00/8 packets on the POSTROUTING chain.

`[ 3417.984315] IN= OUT=tun0 SRC=fe80:0000:0000:0000:cef3:1fd3:93b9:f250 DST=ff02:0000:0000:0000:0000:0000:0000:00fb LEN=92 TC=0 HOPLIMIT=1 FLOWLBL=292378 PROTO=UDP SPT=59920 DPT=5353 LEN=52 `

darkdrgn2k, 2018-05-02:
Also posted on Raspberry Pi forum in case some one out there knows anything
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=212533

darkdrgn2k, 2018-05-02:
Configured a usb dongle to be AP (swapped out driver to make the dongle be the AP) and everything worked as expected.

This indicates a driver/hardware issue?

makeworld-the-better-one, 2018-08-23:
What are the practical implications of this?

darkdrgn2k, 2018-08-23:
Needs to be re-tested

Onboard wifi as ap will connect and allow you to access local resources, but will not allow you to access internet/hyperborea

makeworld-the-better-one, 2018-08-23:
So that's why you wanted me to test the AP... I'll see about doing it tonight on my own, worst case at the meeting.

jedahan, 2018-08-23:
Is it possible there is a sysctl flat that needs to be used like the ipv4 packet forwarding one but for ipv6? Maybe it's interface or more likely driver specific (bcm disabled by default, nl80211 enabled by default)?

darkdrgn2k, 2018-08-23:
Packet forwarding was checked, and the weird part is that a USB dongle works fine, just on board.

Don't even know if its work wasting brain cells on it since its been some time and new OS has been released.  Needs to be re-tested.

darkdrgn2k, 2018-09-01:
Tested today issue persists.

- Internet (ipv4) works fine 
- Hyperboria (ipv6 fc) does not work

Confirmed
```
pi@tomesh-2c05:~ $ cat   /etc/sysctl.conf | grep ipv6
net.ipv6.conf.all.forwarding=1
#net.ipv6.conf.all.accept_redirects = 0
#net.ipv6.conf.all.accept_source_route = 0
```

darkdrgn2k, 2019-03-07:
seems to be working

closed 2019-03-07 by darkdrgn2k