tomeshnet/prototype-cjdns-pi #152: Error when creating ad-hoc interface: command failed: Device or resource busy (-16)

suiluj, 2018-05-04:
I get the following error when trying to create an ad-hoc interface:

```
pi@rpizero1:~ $ set -e
pi@rpizero1:~ $ sudo iw reg set DE
pi@rpizero1:~ $ mesh_dev="wlan0"
pi@rpizero1:~ $ sudo ifconfig $mesh_dev down
pi@rpizero1:~ $ sudo iw $mesh_dev set type ibss
command failed: Device or resource busy (-16)
```

I am using a Raspberry Pi Zero W(H) with the latest Raspbian Stretch 2018-04-18

Does someone have a solution?
I am not sure which process or something is blocking the interface and already tried to stop wpa_supplicant and dhcpcd. But i was not successful.

darkdrgn2k, 2018-05-06:
This is using the on board  raspberry pi zero wireless correct? or an external one.

I will dig up my rpi zero  and try to re-produce it when i have a moment!

suiluj, 2018-05-06:
Thank you @darkdrgn2k 
Yes i am using the internal wifi of the rpi zero w

in my first comment i only tried the commands as shown.

Now i started an installation with the script command:
`wget https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi/master/scripts/install && chmod +x install && WITH_AD_HOC=true WITH_PROMETHEUS_NODE_EXPORTER=true WITH_PROMETHEUS_SERVER=true WITH_GRAFANA=true ./install`

first outputs:
```
pi@rpizero1:~ $ wget https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi/master/scripts/install && chmod +x install && WITH_AD_HOC=true WITH_PROMETHEUS_NODE_EXPORTER=true WITH_PROMETHEUS_SERVER=true WITH_GRAFANA=true ./install
--2018-05-06 18:59:25--  https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi/master/scripts/install
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.12.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.12.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1403 (1.4K) [text/plain]
Saving to: ‘install.2’

install.2                                         100%[===========================================================================================================>]   1.37K  --.-KB/s    in 0s      

2018-05-06 18:59:25 (3.70 MB/s) - ‘install.2’ saved [1403/1403]

Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
Already on 'master'
Your branch is up-to-date with 'origin/master'.
Starting installation on Raspberry Pi Zero (9000c1)...
Configure Mesh Point interface (Y/n)? n
Mesh Point interface configuration will be skipped
Configure Basic Firewall (Y/n)? n
Internet Firewall will be skipped
Configure Internet gateway (Y/n)? Y
Internet gateway will be configured
Install IPFS (Y/n)? n
IPFS installation will be skipped
Configure Hyperboria-compatible Domain Name Servers (Y/n)? n
DNS settings will be skipped
Configure Hyperboria-compatible Network Time Protocol server (Y/n)? n
NTP settings will be skipped
Install non-essential tools useful for network analysis (Y/n)? n
Extra tools will be skipped
Enter the name of your mesh network (default: tomesh): tomesh
```

ending with an error:

```
net/Benchmark.c:191 TODO(cjd): this will fail with a different encoding scheme
wire/PFChan.h:288 * TODO(cjd): This doesn't cover all control message types yet.
benc/serialization/json/JsonBencSerializer.c:331 FIXME(gerard): silently skipping anything we don't understand
client/Configurator.c:573 * FIXME(sdg): moving noforks after setuser might make nproc <- 0,0 work
admin/angel/Core.c:81 TODO(cjd): we need to begin detecting MTU and informing the OS properly!
memory/Allocator.c:246 TODO(cjd): make this optional.
./interface/tuntap/test/TUNInterface_ipv4_root_test.c:59 TODO(cjd): fix TUNConfigurator_addIp4Address() for Illumos, BSD.
dht/dhtcore/NodeStore.h:160 * FIXME(arceliar): Documentation is out of date
./tunnel/test/IpTunnel_test.c:389 Allocator_free(alloc); //TODO(cjd): This is caused by an allocator bug.
./test/Beacon_test.c:16 TODO(cjd): Subnode beacon testing
net/InterfaceController.c:865 TODO(cjd): eps are created in 3 places, there should be a factory function.
switch/SwitchCore.c:19 TODO(cjd): Get rid of NumberCompress so we can set encodingScheme at runtime.
switch/SwitchCore.c:224 TODO(cjd): hmm should we return an error packet?
interface/tuntap/NDPServer.c:128 TODO(cjdns, Kubuxu): Filtering basing on cjdns network and tunnels.
dht/dhtcore/NodeStore.c:669 TODO(cjd): This does not depend on nodeStore or alter the link, consider moving to Node.c
dht/dhtcore/NodeStore.c:900 FIXME(arceliar,cjd): linking every node with 0 link cost, this can't be right.
dht/dhtcore/NodeStore.c:1213 TODO(arceliar): protect keyspace, evict the worst bestParent instead?
dht/dhtcore/NodeStore.c:1293 TODO(arceliar) the 1023 here is pretty arbitrary...
dht/dhtcore/NodeStore.c:1295 TODO(arceliar): is this safe?
dht/dhtcore/NodeStore.c:1523 TODO(cjd): this could return ~0
dht/dhtcore/NodeStore.c:1543 TODO(cjd): this has been the source of assert failures and we might be sweeping
dht/dhtcore/NodeStore.c:1644 TODO(cjd): Schlameil the painter
dht/dhtcore/NodeStore.c:2036 FIXME(arceliar): calcNextCost is guessing what the cost would stabilize to
dht/dhtcore/NodeStore.c:2055 TODO(cjd): What we really should be doing here is storing this link in a
dht/DHTModuleRegistry.c:64 TODO(cjd): Call a debugger with all unhandlable messages?
memory/BufferAllocator.c:25 * TODO(cjd): addOnFreeJob adds a job which is only run when the root allocator is freed
dht/dhtcore/Janitor.c:240 FIXME(arceliar): Probably need stronger requirements than this.
dht/dhtcore/Janitor.c:399 FIXME(arceliar): This target probably isn't optimal.
dht/dhtcore/RouterModule.c:85 * TODO(cjd): ---
subnode/ReachabilityAnnouncer.c:366 TODO(cjd): else if the peer is dropped...
subnode/ReachabilityAnnouncer.c:525 TODO(cjd): This implies a risk of oscillation wherein there is always a tiny bit of
Failed to build cjdns.
Total build time: 1638222ms.
cp: cannot stat '/opt/cjdns/cjdroute': No such file or directory
pi@rpizero1:~ $ 
```

My primary goal is to create an ad-hoc network. The other parts would be bonus.

Iw dev show the wlan0 interface still as managed:

```
pi@rpizero1:~ $ iw dev
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr 3e:b8:35:40:8d:0e
		type P2P-device
		txpower 31.00 dBm
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:7d:5c:14
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 31.00 dBm
pi@rpizero1:~ $ 
```

darkdrgn2k, 2018-05-06:
I was able to replicate your issue with AdHoc.  Seems it may be a driver/firmware issue:

`[   48.960444] brcmfmac: brcmf_cfg80211_change_iface: iface validation failed: err=-16` 

I tried rpi upgrade but that didn't help either
I will dig deeper into this but I also created an issue in the [firmware](https://github.com/raspberrypi/firmware/issues/990) repo for raspberry pi

As for the script/compile

Seems Raspberry Pi Zero may be suffering from the same compile issue  as the Pi 3. Not sure.
I ran out of time waiting for compiling to finish to replicate your issue so ill check back later.

darkdrgn2k, 2018-05-07:
I have modified the compile for Zero to use same method we used in the raspberry pi 3 issue seems to compile fine.

When installing make sure you add TAG_PROTOTYPE_CJDNS_PI=v0.2.1 
IE
`
wget https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi/master/scripts/install && chmod +x install && WITH_AD_HOC=true WITH_PROMETHEUS_NODE_EXPORTER=true WITH_PROMETHEUS_SERVER=true WITH_GRAFANA=true TAG_PROTOTYPE_CJDNS_PI=v0.2.1  ./install
`

*NOTE* the raspberry pi zero runs an overclocked Raspberry Pi 1 processor chip. You can probably get 10-13Mbps out of it running CJDNS

darkdrgn2k, 2018-05-08:
I think this may be a bug in the new Raspbian release.

Try this base image from 2018-03-13

https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-03-14/2018-03-13-raspbian-stretch-lite.zip

NukeManDan, 2018-09-04:
I am actively trying to get: 2x Pi3b+ , 1x 3B , and possibly a fer Zeros on an ad-hoc wireless network. (as internal radios don't do mesh point. :-1: 

https://github.com/althea-mesh/althea_rs/issues/248

Will I need that `TAG_PROTOTYPE_CJDNS_PI=v0.2.1` for all devices? Or just the zeros?

NukeManDan, 2018-09-04:
I get:
```sh
$ WITH_AD_HOC=true WITH_PROMETHEUS_NODE_EXPORTER=true WITH_PROMETHEUS_SERVER=true WITH_GRAFANA=true TAG_PROTOTYPE_CJDNS_PI=v0.2.1 ./install
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://deb.nodesource.com/node_8.x stretch InRelease
Reading package lists... Done
error: pathspec 'v0.2.1' did not match any file(s) known to git.
```

NukeManDan, 2018-09-04:
Now on a RPi3B with fresh flash to this: https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-03-14/2018-03-13-raspbian-stretch-lite.zip

And a clean reinstall with all default values except mesh = N , AP = N , Ad-Hoc = Y:

```This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

   _                           _     
  | |_ ___  _ __ ___   ___ ___| |__  
  | __/ _ \| '_ ` _ \ / _ / __| '_ \ 
  | || (_) | | | | | |  __\__ | | | |
   \__\___/|_| |_| |_|\___|___|_| |_|
---------------------------------------
cjdns Service ................ [ACTIVE]
Ad-hoc Interface ........... [INACTIVE]
IPFS Service ................. [ACTIVE]
ScuttleBot Service ........... [ACTIVE]
Prometheus Node Exporter ..... [ACTIVE]
Prometheus Server ............ [ACTIVE]
Grafana ...................... [ACTIVE]
yrd .......................... [ACTIVE]
---------------------------------------
NODE
xxxxx
---------------------------------------
PEERS
---------------------------------------

...
$ ip a
...
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
...
```

After that - same errors that @suiluj  saw in the original `iw $mesh_dev set type ibss`  and `iw dev` calls.

@darkdrgn2k  got some thoughts on where to start? :smile:

darkdrgn2k, 2018-12-02:
So it seems that we got the raspberry pi onboard ibss to work.

See bottom of https://github.com/tomeshnet/prototype-cjdns-pi/blob/develop/README.md

darkdrgn2k, 2018-12-02:
not exactly sure what you need batman adv

benhylau, 2018-12-14:
Is this ticket still relevant? Can we close? @suiluj @darkdrgn2k

closed 2018-12-15 by suiluj