tomeshnet/prototype-cjdns-pi #164: Find better way to kill procses

darkdrgn2k, 2018-07-23:
Some services ExecStop may be killing process they should not that have the name in them

IE:
`ExecStop=/usr/bin/pkill -f ipfs`
 may be killing ipfs user running bash

jedahan, 2018-07-23:
`ExecStop=/bin/kill -s QUIT $MAINPID` should work, if systemd can guess the MAINPID which it should by default.

jedahan, 2018-07-23:
https://github.com/genenetwork/gn-deploy-servers/blob/692a87595775e54a895ce16f5f25711f3d268a4a/masterfiles/ipfs/ipfs.service as an example, and manual https://www.freedesktop.org/software/systemd/man/systemd.service.html

darkdrgn2k, 2018-07-25:
Found lines that require fixes 

- https://github.com/tomeshnet/prototype-cjdns-pi/blob/8ca9a6ce32c2a21b2d5a4c67f42e112faca9178f/scripts/ipfs/ipfs.service#L10

- https://github.com/tomeshnet/prototype-cjdns-pi/blob/9a819d573ab1adf15213a489d4d6e2c4ec6fdb4e/scripts/prometheus-node-exporter/prometheus-node-exporter.service#L9

- https://github.com/tomeshnet/prototype-cjdns-pi/blob/8ca9a6ce32c2a21b2d5a4c67f42e112faca9178f/scripts/prometheus-server/prometheus-server.service#L10

darkdrgn2k, 2018-07-25:
Also in the ipfs stream repo

- https://github.com/tomeshnet/ipfs-live-streaming/blob/83298ee8ffbf532e16216c988364bb106a028100/terraform/ipfs-server/process-stream.service#L11

- https://github.com/tomeshnet/ipfs-live-streaming/blob/83298ee8ffbf532e16216c988364bb106a028100/terraform/ipfs-mirror/ipfs.service#L11

- https://github.com/tomeshnet/ipfs-live-streaming/blob/83298ee8ffbf532e16216c988364bb106a028100/terraform/ipfs-server/ipfs.service#L11

- https://github.com/tomeshnet/ipfs-live-streaming/blob/83298ee8ffbf532e16216c988364bb106a028100/terraform/ipfs-mirror/ipfs-pin.service#L11

closed 2018-08-16 by darkdrgn2k