tomeshnet/prototype-cjdns-pi #173: Node Exporter and NodeInfo

darkdrgn2k, 2018-07-29:
- What do we would like nodeexporter (or nodeexporter replacement) to report.
- What kind of information do we want nodeinfo.json to provide
- syntax of nodeinfo.json -  [refrence](https://github.com/hyperboria/docs/blob/master/cjdns/nodeinfo-json.md)

darkdrgn2k, 2018-07-29:
C/P From PRs

benhylau:
---
Making cjdns first class makes it difficult to add another router / identity (e.g. yggdrasil, libp2p) in the future. Perhaps:

```
"routers": {
    "cjdns": {
        "key": "__KEY__"
    }
}
```

We don't have to address this now, since this is prototype and we can make breaking changes. Perhaps we need to version this JSON though.

benhylau: 
---
It's better to have stable keys for validation and navigation purposes. Like:

```
"services": [
    {
        "name": "ipfs",
        "version": "__IPFS_VERSION__",
        "params": [__IPFS_PARAMS__]
    }
]
```

This is a very rushed schema design. My take is it will completely change in the future, so stick a version on the schema.

makeworld-the-better-one, 2018-07-30:
I feel like nodeexporter should handle live stats, and the tracking of those stats over periods of time, etc. Nodeinfo should hold mostly static information that's important to have accessible, like the software version nodes are running, or their IPFS ids.

benhylau, 2018-07-31:
I did not realize there is a somewhat official spec. My comments are consistent with the official spec. The official one looks reasonable if it adds a `"version"` field, I don't know why we decided to implement this other schema.

makeworld-the-better-one, 2018-07-31:
@benhylau I agree with you that sticking to the spec is best. I didn't do it at first, but I see how it makes more sense now. And yes a version field would be great as well.

darkdrgn2k, 2018-08-13:
Q: Is exposing IPFS ID to the world a smart idea? Are there security implications to this?

makeworld-the-better-one, 2018-08-13:
I believe IPFS nodes exchange ids anyway, verifying that they match public keys and things. I don't think it should be an issue.

makeworld-the-better-one, 2018-08-13:
Yes I believe that may be possible as well, but it's still necessary. Cause to add a node as bootstrap you need that ID anyway, and you can't connect to it without the ID, so you can't probe any handshake.

benhylau, 2018-08-13:
I don't think it is an issue to expose IPFS ID. They are publicly shared already. However, I think it's important that we don't end up with stale info on the json.

@darkdrgn2k can you explain more what you mean by probing the handshake?

What is the purpose of knowing the IPFS of a node in the first place?

darkdrgn2k, 2018-08-13:
> There really should be a shared module that handles schema version checking and parsing, instead of having each module jq parse this. In the current model, it means once the schema updates, every module will need to update its nodeinfo parsing.