tomeshnet/prototype-cjdns-pi #175: nodejs using udev rules

darkdrgn2k, 2018-08-05:
Possible issues on some devices running nodejs (cexec) via udev

https://github.com/nodejs/node/issues/19036

https://github.com/nodejs/node/issues/16901

benhylau, 2018-08-21:
How may this affect us? The cjdns compilation?

darkdrgn2k, 2018-08-21:
@ASoTNetworks ran into this bug on a computer (not SBC) when trying to create an exit node using iptunnel.

`cexec` uses nodejs, which had an issue executing. He found the above mentioned bugs.

Script not executing properly
https://github.com/tomeshnet/prototype-cjdns-pi/blob/master/scripts/cjdns-iptunnel/cjdns-setup

Running it manually after the fact brought the configuration up without a problem just not in the udev environment

darkdrgn2k, 2018-10-14:
On debian node crashes when run in udev env

```
root@exit1:/home/user# nodejs --version
v8.12.0
```
```
[1907700.743443] node[11729]: segfault at f ip 0000000000fe8280 sp 00007ffd8f3d72a0 error 6 in node[400000+1b60000]
```
works fine when run out of  uden env

darkdrgn2k, 2018-10-14:
possible work around

Add this 
`ExecStartPost=/usr/local/sbin/cjdns-setup`
to

`/etc/systemd/system/cjdns.service`

then run
 `sudo systemctl daemon-reload`

closed 2019-01-28 by darkdrgn2k