tomeshnet/prototype-cjdns-pi #181: Missing DNS servers for AP

darkdrgn2k, 2018-08-17:
When NO source of DNS servers is available, IE
 - Device is not connected to a network that provides DNS server ip
 - Device does not use hyperborea dns when hyperborea is available

There is no DNS resolution for that device/access point even when IP TUNNEL is used to get Internet.

makeworld-the-better-one, 2018-08-17:
>Device is not connected to a network that provides DNS server ip

How likely is that? I understand we need to prepare for it, but what's an example of when that would happen?

>Device does not use hyperborea dns when hyperborea is available

How could that happen? Doesn't our code specify to always do that?

>There is no DNS resolution for that device/access point even when IP TUNNEL is used to get Internet.

So can the solution be to add a clearnet last resort resolver, to be used over IP TUNNEL if necessary? Or you're saying that even if you do that it still doesn't work?

darkdrgn2k, 2018-08-17:
This is the first time this has happened/been reported however it is a total valid setup for the pis. The issue is when there is no source for a dns server (because dhcp is not used)

Yes the solution is easy and would be to add a default clear net resolver. The work around I provided the person was to add `server=8.8.8.8` to the bottom of `/etc/dnsmasq.conf`  Simple enough fix

However the question becomes. Use 8.8.8.8 or maybe 192.99.85.244 or what dns?

Issue was logged as a reminder that task needs to be completed.

jedahan, 2018-08-17:
Maybe an [OSRN dns](https://en.wikipedia.org/wiki/Open_Root_Server_Network)?

I'd prefer OSRN over cloudflare 1.1.1.1 over google 8.8.8.8

benhylau, 2018-08-29:
Is the intention to use OSRN? Currently we are on Cloudflare:
https://github.com/tomeshnet/prototype-cjdns-pi/blob/v0.3/scripts/hostapd/install#L13-L14

This is not critical to the release but https://github.com/tomeshnet/prototype-cjdns-pi/pull/186/commits/fb71f6f31a00634f0c7a8a85e2446ed41dd98504 is incorrect and we should either close this issue if we want to stay with 1.1.1.1 or keep this open if the decision is to use OSRN.

darkdrgn2k, 2018-08-29:
Ah i miss understood it as  OSRN on top of CLOUD FLARE not OSRN vs CLOUDFLARE.

Not sure what OSRN servers even are or how stable they are.

Input welcome

darkdrgn2k, 2018-08-29:
Ok so the options so far are 

Google 8.8.8.8 👎 
Cloud Flare 1.1.1.1
OSRN (https://en.wikipedia.org/wiki/Open_Root_Server_Network)
OpenNic (https://www.opennic.org/)

makeworld-the-better-one, 2018-08-29:
It's worth noting that OSRN has no servers nearby, they are all in Europe, with one in India. I haven't had the chance to personally test response time, but closer might be better. OpenNic has some closer ones, but they are less reliable because they are community run. Google doesn't respect your privacy, and while Cloudflare says they do, people are starting to distrust even them now. I personally think that Cloudflare is our best bet at this point, for speed and privacy. However if we can guarantee the reliability of a nearby OpenNic server somehow, that might be better. It's also worth considering that Cloudflare has more world wide servers than some of these other options, and people from all over the world have been using this.

jedahan, 2018-08-30:
I know performance matters, but is DNS latency really the system bottleneck? 

I ran namebench from my home computer `docker run --net=host anroots/namebench -O 212.224.71.71 1.1.1.1 8.8.8.8 192.99.85.244` (ORSN Cloudflare Google OpenNIC) and got the following:

```
Mean response (in milliseconds):
--------------------------------
1.1.1.1          ######### 47.93
8.8.8.8          ############## 78.89
212.224.71.71    ################################ 176.90
192.99.85.244    ##################################################### 300.31
```

I'm gonna run another test with 4 different ORSN and 4 different OpenNIC to see if there is high variability, but generally we can say ORSN is about 2x slower than Google, and OpenNIC is about 4x slower than google.

jedahan, 2018-08-30:
```
Mean response (in milliseconds):
--------------------------------
1.1.1.1          ######## 59.93
8.8.8.8          ############# 97.74
172.98.193.42    ####################### 183.92
212.224.71.71    ########################## 206.55
18.211.225.60    ########################### 208.42
37.187.99.178    ############################# 226.58
192.99.85.244    ############################################ 345.94
162.248.241.94   ############################################### 370.61
163.172.48.48    ################################################# 392.17
79.133.62.62     ##################################################### 424.64
```

makeworld-the-better-one, 2018-08-30:
>I know performance matters, but is DNS latency really the system bottleneck?

I have no idea. I just assumed it was something to consider. 

Are you testing from Toronto?

darkdrgn2k, 2018-08-30:
DNS is a unique beast.

First if its not working, people consider the internet DOWN.
Both Google and Cloud flare use ANYCAST ips, you are routed the the closest ip address to you
Also they have multiple servers that answer for that ip address so if one goes down another takes its place

Using multiple addresses on the client side just round robins the requests. Its not if one is down go to another. 

Response lag does not sounds like an important metric, but the "standard" is to have a website in front of the end user in 3 seconds, any longer you risk them getting impatient. There are only so many things you can do to shave of as much time as you can and although 100 milliseconds does not sound like much its 100 milliseconds PER domain resolution and many website has google analytics, facebook etc etc links in their site.  It adds up fast.

Not saying this is a make it or break it thing at all, but things to consider. Especially if you layer on top of all this mesh latency. But having a reliable dns server cuts down the time of debugging.

benhylau, 2018-08-31:
I definitely choose reliability on DNS. After HOPE when I was prepping the dweb demos, I ran into so many DNS issues at my friend's house, and wasted hours because some software repository didn't resolve properly, until eventually I changed to `8.8.8.8`.

I also think we are testing a lot of moving parts here with the mesh routing already. I rather have the fastest and most reliable DNS service than to have latency and have to think about whether it's the DNS or the router. Also we will eventually need a mesh DNS solution, this DNS is just a transitional dependency. My vote is `1.1.1.1`.