tomeshnet/prototype-cjdns-pi #196: Better documentation for modules

darkdrgn2k, 2018-09-01:
# Proposal

Create a files (maybe modules.md) that lists all the modules, a short description of what they do, if applicable how to use them.

thoughts?

jedahan, 2018-09-01:
Maybe start in the wiki to make things easier to collaborate on, then submit a PR either as readme.md for modules that need explaining, and a one-liner in the main repo?

See geometry plugins as an example:

https://github.com/geometry-zsh/geometry/tree/master/plugins

https://github.com/geometry-zsh/geometry/tree/master/plugins/exec_time

makeworld-the-better-one, 2018-09-01:
This would possibly replace the sort of explanation going on in the readme.

benhylau, 2018-09-01:
I feel that a single `MODULES.md` linked to from `README.md` is sufficient since our plugins do not require much explaining. Scattering them make them harder to navigate and the overhead to create a new module becomes higher.

I also don't want to start the Wiki, because it's publicly visible and having two places (wiki and repo files) out of sync make it confusing for users.

darkdrgn2k, 2019-02-12:
Basic modules framework created

Needs more update

closed 2019-03-04 by darkdrgn2k