tomeshnet/prototype-cjdns-pi #201: shellcheck integration?

jedahan, 2018-09-01:
https://github.com/koalaman/shellcheck/ is pretty nice

I've only ever had to avoid a single instance of a warning it gave me.

It helps move towards consistent programming style, and I've learned some gotchas along the way.

It is built into travis CI so if that got enabled on pull requests, could be cool.

benhylau, 2018-09-01:
Yes please!

darkdrgn2k, 2018-09-03:
Ha didnt see this issue :) i was working on it this week on #203 glad to see we had the same idea :)
I already came across on error i needed to avoid but thats ok!

Ok I cleaned up a big chunk of the script as per shellcheck suggestions
Preliminary integration with travis as well

Its all a work in progress suggestions welcome!
Especially how to fix this one :)
```
In ssb-web/install line 8:
source "$BASE_DIR/../shared/nginx/install"
^-- SC1090: Can't follow non-constant source. Use a directive to specify location.
```

jedahan, 2018-09-03:
all shellcheck errors I've seen have a corresponding wiki page

https://github.com/koalaman/shellcheck/wiki/SC1090

So I think from that add a directive with the full expected path
`# shellcheck source=../shared/nginx/install` ?

darkdrgn2k, 2018-09-03:
That seems like allot of work for a check that should not really fail :/ maybe i will just disable SC1090 then. Do you know if you have to disable it in every file or is there a global disable you can do via command prompt.

Also i managed to get most of the error corrected, not quite sure how to to the ones that are left.

darkdrgn2k, 2018-09-03:
`shellcheck --exclude SC1090 `

darkdrgn2k, 2018-09-21:
Completed with latest develop #203

closed 2018-09-21 by darkdrgn2k