tomeshnet/prototype-cjdns-pi #207: Prometheius Install Line

darkdrgn2k, 2018-09-09:
This command line works better to start the install of the prototype node.

`bash <(curl -s https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi/master/scripts/install)`

vs

`wget https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi/master/scripts/install && chmod +x install && ./install`

Benefits

- Fewer commands
- Shorter
- Does not save the install file (subsequent attempts create install.1 install.2 etc
- Does not leave a random "install" file

Thoughts?

jedahan, 2018-09-09:
Might wanna use -L to follow redirects in the curl I think

benhylau, 2018-09-09:
I have used a simpler command before that does not leave an install file. I ran into a weird bug that when installing... I think ipfs, it errors. I think I was doing what nvm does: https://github.com/creationix/nvm#install-script

So I am okay with changing the install command provided that we test properly.