tomeshnet/prototype-cjdns-pi #210: Captive portal (WiFi discoverability for newbie users)

DaniellMesquita, 2018-09-16:
When connecting to WiFi, it shows a captive portal, where user can signup/login, and owner can customize it.

makeworld-the-better-one, 2018-09-16:
>where user can signup/login

Signup or login to what? Some services on the pi may eventually have a login service online, but having a captive portal for the whole pi doesn't make sense. As for allowing wifi access, a WPA password should be enough. EAP is also supported, which allows logins, although we don't have a portal for that at the moment. Feel free to add it for your own use though!

DaniellMesquita, 2018-09-16:
Not good for offline users to can't login WiFi because it haves no password.
Also, when they log on open WiFi, they doesn't knows how to join CJDNS.

DaniellMesquita, 2018-09-16:
@makeworld-the-better-one
Do you have a contact other than GitHub?

benhylau, 2018-09-16:
I also believe a captive portal is out of scope for this project. It is not intended to be a public access point allowing users to sign up new accounts. It's an installer to set up your own mesh node with cjdns and peer to peer apps.

bonfus, 2018-09-17:
What about a third option in `hostapd/install` which provides an open wifi with restrictions on private network addresses (i.e. 10.0.0.0/8, 172.16.0.0/12, 192.168.0.0/16, fd00::/8) ?
I'm drafted it on my installation and would be happy to open a PR if you think it fits the scope of your project.

makeworld-the-better-one, 2018-09-17:
>Not good for offline users to can't login WiFi because it haves no password. Also, when they log on open WiFi, they doesn't knows how to join CJDNS.

@

makeworld-the-better-one, 2018-09-17:
>Not good for offline users to can't login WiFi because it haves no password. Also, when they log on open WiFi, they doesn't knows how to join CJDNS.

@DaniellMesquita

Joining CJDNS isn't really part of the code, it only provides *access* to CJDNS. This happens automatically so you don't need to worry about users. 

If I understand correctly, your other concern is that if your WiFi has no password, users won't be able to have a login. I would reccomend a password for security, but if you don't have one, then everyone can still use your WiFi fine. The issue for you is with creating an account I think, which I am confused about. Why?

DaniellMesquita, 2018-09-17:
@bonfus

> What about a third option in hostapd/install which provides an open wifi with restrictions on private network addresses (i.e. 10.0.0.0/8, 172.16.0.0/12, 192.168.0.0/16, fd00::/8) ?
> I'm drafted it on my installation and would be happy to open a PR if you think it fits the scope of your project.

It is the only form to make it discoverable by newbie/offline users, for the network's growth.

benhylau, 2018-09-18:
@bonfus is it possible to have this as a contrib script completely isolated from main install flow? I am concerned about long term maintainability of this extra option, and prefer it to not affect main install flow.

bonfus, 2018-09-18:
That makes sense. I'll put it in a github gist that can installed (and uninstalled) after a successful setup.

benhylau, 2018-09-18:
We have a contrib folder :)

https://github.com/tomeshnet/prototype-cjdns-pi/tree/master/contrib

@darkdrgn2k how do you feel about having the script here?

darkdrgn2k, 2018-09-18:
We have discussed captive portals at length and never came to a conclusion.

Although I do agree that it would be nice to inform a passer-by what this craziness is, it also goes against the whole "having a "free"and open network". Captive portals is they hijack connection, break SSL and bunch of other stuff that one can argue break the spirit of that.

One alternative approach was the name the SSID with a website that would explain it (is "Visit tomesh.org")

Another one we talked about is using [nxdomain](https://www.dnsknowledge.com/whatis/nxdomain-non-existent-domain-2/) but that would require CJDNS only DNS servers which could be a problem too.

We also discussed a crazy idea of dual SSID's one "captive portal" to would only allow you to generate a EAP/certificate, and a second that is free and open using the EAP/Certificate.  That would protect the user even more as each connection would be uniquely encrypted.

I think a contrib script is fine, at very least would let us experiment with the concept more then just theorizing. 

I think this would be an interesting conversation to take place when/if we ever do a UI for "configuration" or something if we ever get there.

darkdrgn2k, 2018-09-18:
@DaniellMesquita 

PS: We usually sit on matrix (http://chat.tomesh.net or on irc efnet #tomesh) if you would like a longer debate/conversation/etc about this stuff.

DaniellMesquita, 2018-09-18:
> One alternative approach was the name the SSID with a website that would explain it (is "Visit tomesh.org")

But if users are offline? Mesh networks haves no sense if not fully support offline users.

darkdrgn2k, 2018-09-18:
This is a bigger question now what defines "OFFLINE" and an even bigger question of what does the node actually supposed to do/provide.

DaniellMesquita, 2018-09-18:
> We also discussed a crazy idea of dual SSID's one "captive portal" to would only allow you to generate a EAP/certificate, and a second that is free and open using the EAP/Certificate. That would protect the user even more as each connection would be uniquely encrypted.

Dual SSIDS will not make users confused? How the first SSID will tell that another is needed?
Regarding encrypted connections, [ZeroNet](https://zeronet.io) does it.

darkdrgn2k, 2018-09-18:
There is allot to that idea that would need to be worked out.  I actually got this idea from linkNYC because they have that as an option. There is the "FREE" wifi and then there is the FREE wifi with certificate.

Its also NOT possible with the onbaord rpi wifi since it only supports one SSID

darkdrgn2k, 2018-09-18:
> Regarding encrypted connections, ZeroNet does it.
Not quite sure how ZeroNet does the "encrypted connections". My concern is network level encryption since we cant trust the user to be mindful of their privacy!

CJDNS deals with inter mesh encryption, however the Legacy network (AP-Client) is the issue here

AP's can be configured as following
No Password - Anyone can listen and capture your traffic, application layer encryption needed (is https) - Client needs to do ti
WEP - Basically same as no password
WPA/PSK - If you know the key you can decrypt all the traffic and since your making the WPA key public everyone know is
WPA/EAP - Each connection is individually encrypted. No public keys needed just Login/Password or Signed Certificate
WPA3 ??????? Maybe ????

DaniellMesquita, 2018-09-19:
But the SSID needs to not be confused, and discoverable/intuitive.

darkdrgn2k, 2018-09-19:
SSIDs can be made intuitive
`tomesh-Start Here`

benhylau, 2018-09-19:
I agree with what @darkdrgn2k said about captive portal and am personally opposed to use of captive  portals. It also makes certain use cases difficult (e.g. connect a router to it and use the link as backhaul).

Also agree with @DaniellMesquita about how the dual SSID is terribly confusing for user.

It seems that EAP is the preferred way to connect to a public hotspot. I think it is possible to use wildcard login details (i.e. accept any user and password). Why don't we set that up, and in the SSID do something like `free-35df-eap-any-user-pass`?

I understand that WPA is more familiar to users but it seems EAP is necessary for security purpose anyways. From what I understand there is no need to generate new certs per user.

benhylau, 2018-09-19:
To clarify, I am still fine with having a `captive-portal` in contrib, as much as I am not a fan of it. This is a prototype repo and if people want a feature enough to send a PR for it we generally accept it, provided it doesn't add long term maintainance burdens and meet quality standards.

darkdrgn2k, 2018-11-13:
>I understand that WPA is more familiar to users but it seems EAP is necessary for security purpose anyways.

EAP (as awesome as it is) actually makes it very difficult for end users to use. 

Mac seems to be just a warning you can ignore

On Android the fields are a little confusing (Username, Anonymous Identify, Password) which anonymous identify needs to be empty

Newer versions of android need to select another option to ignore untrusted CAs

Windows has it the worst.  You need to MANUALLY configure the access point you want to connect to because there is a check box you need to select to ignore untrusted CAs.  

Even using trusted CAs you need one from a root certificate authority since you cant use intermediate certificates.

That's why i always suggest using WPA for first time users because its much easier to connect to.

darkdrgn2k, 2018-11-13:
Also regarding Captive Portals: You may not want a captive portal for your purpose. Captive portals allow a user to "solve" somehow the captive portal (buy a account, pay for it, click an i accept button) and be released back into the normal network.

What your looking for is just a small subset of what a captive portal does -  redirect user to a webpage.

Something like
`iptables -t nat -A PREROUTING -p tcp --dport 80  -j DNAT --to-destination 10.0.0.1:80`

That redirects ALL traffic going to any port 80 on ipv6 to a new ip address (10.0.0.1 being the pi) port 80.
This way any NON SSL website you request will show you the "captive portal" page (but not be a captive portal)

All other traffic will still work as normal (on a closed system go no where)
This model makes it easy to work around things like back-hauls since its not a heavy captive portal backend but a simple iptables line.

darkdrgn2k, 2018-11-13:
> dual SSID is terribly confusing for user.

Onboard rpi cant do dual SSID anyway.

LinkNYC has multiple SSIDs

Open clear one  - anyone can go on and use it
Certificate encrypted EAP - anyone can register on the open clear one and download a certificate for themselves to use on the EAP network

So in this model the uninitiated have an easy access to the network while the ones who wish to participate in the encrypted network can.

darkdrgn2k, 2018-11-23:
https://github.com/tomeshnet/prototype-cjdns-pi/blob/offline/contrib/offline/install.sh

Script to change the WiFi access point to a "captive portal" style by redirecting all traffic being sent to  80 to 10.0.0.1

Disables masquerading ipv4 (allowing hyperboria to still work) and configured nginx to redirect 404 errors to /

darkdrgn2k, 2018-12-14:
@DaniellMesquita can we close this issue or we still want to peruse a real captive portal?

closed 2019-01-06 by darkdrgn2k

DaniellMesquita, 2019-01-06:
@darkdrgn2k

Thanks for believing on that, seeing that potential and for working on it; your efforts are very helpful for a decentralized Internet.
How these captive portals works? Is there an API to disable captive portal for specific users after they doing an action?

darkdrgn2k, 2019-01-07:
Details on how this works can be found here
https://github.com/tomeshnet/prototype-cjdns-pi/pull/229#issuecomment-443211299

However
This is *NOT* a "captive portal" in the traditional sense.  There is no way to "solve" the "Captive Portal".

It is a simple re-directs ALL traffic going to a port 80 IPv4 address to the local nginx server instead. By default it will be a "status" page for the node advertising the resources it has access to. It is meant to help educate people of what exactly they connected to.

It does NOT block or interfere with ANY other traffic. Meaning HTTPS or other protocols will not be affected. However it DOES remove masquerading (NAT) of IPv4 traffic so if your behind a firewall TCP traffic will work anyway (out to the internet). 

It is assumption is that this is an "offline", and only has access to Hyperboria.

Hyperboria traffic will not be affected so http://h.chat.tomesh.net will still work (if your on hyperboria)

NOTE: if you do some iptables magic you can manually NAT certain IPs/MAC addresses while blocking the rest but its a manual thing.