tomeshnet/prototype-cjdns-pi #219: cjdns-iptunnel does not work unless hostapd side effects occurred

benhylau, 2018-10-14:
To get cjdns-iptunnel working, we need to enable packet forwarding and NAT:

```
sudo sed -i 's/#net.ipv4.ip_forward=1/net.ipv4.ip_forward=1/' /etc/sysctl.conf
sudo sed -i 's/#net.ipv6.conf.all.forwarding=1/net.ipv6.conf.all.forwarding=1/' /etc/sysctl.conf

iptables -t nat -A POSTROUTING -o YOUR_INTERNET_IFACE -j MASQUERADE
```

These were done by installing hostapd on Raspberry Pis. In my case, it's a VM without an AP, but this also means if we install a Pi as an exit, without using the hostapd feature, the cjdns-iptunnel feature will not work.

I am thinking to add the first two lines to `scripts/cjdns-iptunnel/install` and the iptables line to `scripts/cjdns-iptunnel/cjdns-setup`. The first two lines are idempotent but the last one shouldn't be run if hostapd already did it right? Is there a better place to do these?

benhylau, 2018-10-14:
In the case of the VM, your fixes here is also needed https://github.com/tomeshnet/prototype-cjdns-pi/issues/175

darkdrgn2k, 2018-10-14:
I was thinking about how we put too much weight on what happens in hostapd. I think we should look at maybe enabling forwarding regardless since there are so many permutations of different modules that could use it. Option two is enable them in every module.

Natting would have happened at cjdns-setup but was dieing because of the -e

Ref: https://github.com/tomeshnet/prototype-cjdns-pi/blob/d73e61268ab06825c13eb19d1ccaec3bce69072a/scripts/cjdns-iptunnel/cjdns-setup#L38

You may want to add a sleep at the top cause there is a bit of a race condition where cjdns may not have the tun0 up by the time you are ready.

However we may be able to also get away with using tun+ instead of tun0 to identify any tun interface.

darkdrgn2k, 2018-11-13:
Corrected in cjdns-setup

https://github.com/tomeshnet/prototype-cjdns-pi/blob/cjdns-iptunnel-ipv6/scripts/cjdns-iptunnel/cjdns-setup#L9

benhylau, 2018-11-15:
I agree with all you said, and this option:
>enabling forwarding regardless

closed 2019-03-03 by darkdrgn2k