tomeshnet/prototype-cjdns-pi #248: Nginx installation

darkdrgn2k, 2018-12-01:
nginx right now installes for IPFS and some SSB modules

Should it always be installed? It serves.
 - Welcome page
 - nodeinfo.json

also

- [ ] Add check if its installed before trying to write to it

makeworld-the-better-one, 2018-12-01:
I would say it should, especially with the new welcome page. It wouldn't be uncommon for someone not to install IPFS and SSB, but to want be able to see the welcome page with the node graphs. And nodeinfo.json is helpful.

benhylau, 2018-12-02:
I think nginx should be default installation

darkdrgn2k, 2018-12-26:
C/P from https://github.com/tomeshnet/prototype-cjdns-pi/pull/252#discussion_r243927639

@benhylau 
So which are we doing?

    nginx in shared
    classify into default vs. optional
    Leave nginx in root like cjdns

makeworld-the-better-one, 2018-12-26:
I would get rid of `shared`, and move `nodeinfo` out of it, switching to an `optional` vs `default` setup. I also think CJDNS should be moved out of the script and into it's own module, in the `default` folder.

PR #252 (2018-12-26): Make nginx install by default
Fixes #248. Yay!
closed 2019-01-26 by darkdrgn2k