tomeshnet/prototype-cjdns-pi #257: USB WirelessAC UBS Dongles poor functionality

darkdrgn2k, 2019-01-06:
A previous thought that the AC dongles were under powered and that's what gave them such poor performance/functionality

Found this reference on the internet:
>I just had the idea to set max_usb_current = 1 in /boot/config.txt to make the PI supply as much power as possible to the USB port.

Be worth taking a look to see how well/poorly they preform with this turned on and off

makeworld-the-better-one, 2019-01-06:
> as possible

That scares me. Needs testing I guess, but I worry about other devices, and the pi power, etc. It'd be good to see some documentation for it.

darkdrgn2k, 2019-01-06:
if you can find it please posting

at the. moment it's a hypothesis only

darkdrgn2k, 2019-01-20:
# rtl8812AU

`https://github.com/xxNull-lsk/rtl8812AU.git`
Patch required:  
`sed -i -e "s/__vfs_read/kernel_read/" os_dep/osdep_service.c`
`sed -i -e "s/vfs_read/kernel_read/" core/efuse/rtw_efuse.c`
`sed -i -e "s/vfs_read/kernel_read/" core/rtw_wlan_util.c`

2.4 ghz test
- [x]  Modules Installs
- [x] Enters Adhoc Mode
- [ ] iw reports correctly
- [ ] iw station dump works
- [x] Pings across
- [x] iperf3 60 seconds sustained