tomeshnet/prototype-cjdns-pi #259: Add backup option

darkdrgn2k, 2019-01-20:
Idea - 
Add a "Backup" feature to create file(s) that can be placed on a USB drive, them the SD-Card re-build with the latest prototype build and then data restored to it to allow for easy-er upgrades

What should get backed up

- [ ] /etc/cjdroute.conf
- [ ] /etc/yggdrasil.conf
- [ ] /home/pi/.ssb
- [ ] /home/pi/.ipfs

Though + what to add

makeworld-the-better-one, 2019-01-21:
I like this idea a lot, although I wonder about if it distracts from the upgrade idea of deb pkgs. 

I would also suggest providing the option to back up the home folder entirely.

darkdrgn2k, 2019-01-31:
> I like this idea a lot, although I wonder about if it distracts from the upgrade idea of deb pkgs.

its good if you want to blow up your node and rebuild or even move nodes
and as we all know sd cards fail

makeworld-the-better-one, 2019-01-31:
```
sudo tar uvf /tmp/backup.tar /home/pi/.ssb
sudo tar uvf /tmp/backup.tar /home/pi/.ipfs
```

How's that?