tomeshnet/prototype-cjdns-pi #260: Give AP clients yggdrasil addresses

makeworld-the-better-one, 2019-01-23:
A router running Yggdrasil can give each of its clients a Yggdrasil subnet address, freeing them from being NATed or controlled, and giving each an address on Yggdrasil without needing to install the software. This is a huge use case for this repo. It would require we install `radvd` on the pis, but that's about it. Here's a [guide](https://yggdrasil-network.github.io/configuration.html#advertising-a-prefix).

darkdrgn2k, 2019-01-25:
My test show 
*note* doc say 300 to 200 but in my case it was 301 changes to 201    

We can just change
https://github.com/tomeshnet/prototype-cjdns-pi/blob/d73e61268ab06825c13eb19d1ccaec3bce69072a/scripts/hostapd/radvd.conf#L4

from `prefix fdfc::/64`
To yggdrasil ip address 
ie `prefix 301:a3b4:b826:3cef::/64`

And add 
` ip addr add 301:a3b4:b826:3cef::1/64 dev wlan-ap` to /etc/hostapd/nat.sh 

or more properly configure /etc/network/interfaces with correct ip address

This will preserver cjdns's natting and allow for yggdrasil no-nat  routing

makeworld-the-better-one, 2019-01-25:
Thanks! I have a branch going with this a separate module. Maybe I should stick with thar, so it can be chosen on install, and easily uninstalled as well? Instead of putting it in the install files of other modules.

benhylau, 2019-01-25:
It seems this doesn't need to be its own issue, but need some clarification in the README. Perhaps a short line in that flag table saying if ygg is enabled it will also do this address assignment to AP clients. Also make sure this doesn't mess things up if AP client is disabled.

As for Yggdrasil addresses, I don't understand what you mean, but I think we can only assume `03` and the next 8 bits can be anything. For example, my grep in ipfs live streaming repo for the ygg address is:

`ifconfig ygg0 | grep -E 'inet6 2[0-9a-fA-F]{2}:`

makeworld-the-better-one, 2019-01-25:
What was decided in chat: it won't be it's own module, and the firewall will protect it by default. The README will document how to change the firewall, for a port, all ports of a client, or all clients.

darkdrgn2k, 2019-01-25:
I would not add MORE to readme, its already to big and hard to find things sometimes.

I would preferably like  to see https://github.com/tomeshnet/prototype-cjdns-pi/issues/196 start taking shape.  Maybe include that information in the Yggdrasil module (who would know to look in hostapd) and refrence the hostapd module

darkdrgn2k, 2019-01-25:
> As for Yggdrasil addresses, I don't understand what you mean, but I think we can only assume 03 and the next 8 bits can be anything. For example, my grep in ipfs live streaming repo for the ygg address is:

Guide showed `200:1111:2222:3333:4444:5555:6666:7777` turn into `300:1111:2222:3333::/64`
Didn't specify 200 vs 201. But yes thank you for clarifying. only the first byte (03) changes in the address.

darkdrgn2k, 2019-01-25:
> ifconfig ygg0 | grep -E 'inet6 2[0-9a-fA-F]{2}'

Reference yggdrasil status script. 
@makeworld-the-better-one  if there is abetter way to extrapolating this from yggdrasil we should up date it here to

https://github.com/tomeshnet/prototype-cjdns-pi/blob/d56591c534da9d2d942e1a4bc72f8544125303df/scripts/status#L123

benhylau, 2019-01-25:
This explains it https://yggdrasil-network.github.io/2018/07/28/addressing.html and I also recall asking Arc or Neil when I was figuring this out, to make sure the docs are up to date.

makeworld-the-better-one, 2019-01-25:
I can use this issue/PR to start what's proposed in #196, if that's ok, I like that idea.

darkdrgn2k, 2019-01-25:
> I can use this issue/PR to start what's proposed in #196, if that's ok, I like that idea.

Yes i think thats fine

darkdrgn2k, 2019-01-26:
IP Address for the wlan-ap interface  can be configured by simply appending to `/etc/network/interfaces`

```
iface wlan-ap inet6 static
  address 303:735e:630f:7741::1/64
  netmask 64
```

darkdrgn2k, 2019-01-31:
Implementing in #269

makeworld-the-better-one, 2019-02-17:
Works in develop now.

closed 2019-02-17 by makeworld-the-better-one