tomeshnet/prototype-cjdns-pi #281: Raspberry Stretch: Regular or Full?

ForgivenNin, 2019-02-01:
On my previous question ( #258 ), I asked if there were any other compatible operating systems besides Raspbian Stretch Lite (especially because of the reboot & login problem it has), and you gave me an answer about that.

Before I try setting this up, though, would it matter if I used either the Raspbian Stretch that has recommended software or the one that does not? If it does, which one?

I just don't want to mess up my Pi in the process of doing this.

makeworld-the-better-one, 2019-02-01:
Personally, I just install Raspbian Lite, and enable SSH.

ForgivenNin, 2019-02-01:
Since I have an LCD screen that goes with the Pi, I don't really see a need for SSH.

Would be a good idea if I enabled it, but locked it? I've just heard that when SSH is turned on it opens up security vulnerabilities - especially since SSH is wireless.

makeworld-the-better-one, 2019-02-01:
SSH allows you to enter commands, and access your pi through other computers. It's more useful than using a screen and keyboard, because you don't have to switch computers and things like that. For example, you can easily copy and paste stuff from a browser on your computer into a command to run on the pi. 

Enabling SSH won't be a security risk, as long as you change the password of the pi. You can do that by running the `passwd` command, or by installing our software, which will make you change your password anyway I believe. Pick a strong one and you'll be fine!

ForgivenNin, 2019-02-05:
I was about to close this issue, but I had one last idea/thought of the matter.
(Like don't get me wrong - those were really helpful answers)

What if I used a virtual machine/box for Raspbian Stretch Lite while using something like the actual Raspbian (desktop version)? 
By doing this I could hopefully use my pie for other purposes as well (and not have to get a Raspberry Pi zero or something just solely for connecting to the mesh).

If it turns out to be a little experimental or even problematic, wouldn't all I have to do is close the machine down and things would go back to being fine?

makeworld-the-better-one, 2019-02-05:
No. This will not work. A Raspberry Pi does not have enough resources to run a VM. 

If you want to run our software, it will still work on the desktop version, just know that it is not designed for that, and if you try to do a lot on it the Pi will not like it. Feel free to try it out with desktop version though.

makeworld-the-better-one, 2019-02-05:
I'm going to close this issue, but feel free to still ask questions here or in our chat.

closed 2019-02-05 by makeworld-the-better-one