tomeshnet/prototype-cjdns-pi #282: Hostapd network scan fails

darkdrgn2k, 2019-02-01:
`for wlan in $(find /sys/class/net -maxdepth 1 -name 'wlan*' | xargs -0 basename); do`

line only return first result instead of all of them

makeworld-the-better-one, 2019-02-01:
What is this used for?

darkdrgn2k, 2019-02-01:
To clarify

the folder `/sys/class/net` contains all the network interfaces

`/sys/class/net/$wlan/device/driver` is a SYMLINK to the driver it uses

`readlink` will output the path to the symlink target
`basename` will output the filename in a path yeidling the driver for the interface

currently the for loop should iterate through all WLANx but it only does the first one

makeworld-the-better-one, 2019-02-01:
Can we just disable shellcheck for that area?

darkdrgn2k, 2019-02-02:
we could
but we should probably devise a better line.

Shell check comments are valid.

darkdrgn2k, 2019-02-13:
Interesting read on bash scripting and for loops

https://mywiki.wooledge.org/BashPitfalls#pf1

Simpler loop solution
```
for wlanfile in /sys/class/net/wlan*; do
    wlan="$(basename "$wlanfile")"
```

closed 2019-02-13 by darkdrgn2k