tomeshnet/prototype-cjdns-pi #286: Develop branch install ends abruptly

dasanchez, 2019-02-10:
The installation process ends abruptly with the following line if yggradsil is selected as an option:
```
usermod: group 'yggdrasil' does not exist
```
This was using the _develop_ branch, as of 2019-Feb-9.
```
wget https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi/develop/scripts/install && chmod +x install && TAG_PROTOTYPE_CJDNS_PI=develop ./install
```

[image]

darkdrgn2k, 2019-02-10:
That line was moved into nginx.  

Either i forgot to remove it or it came back in one of the prs.

Thanks for spotting, should be good now.

makeworld-the-better-one, 2019-02-12:
Can this be closed?

dasanchez, 2019-02-12:
@makeworld-the-better-one let's make sure the issue goes away.
I'm reflashing the node now. I tried running the install off develop after running the uninstall script from master and I got this:
[image]

darkdrgn2k, 2019-02-12:
Uninstalls is horrible :( reflash is better ;)

Ok i patched that error. let see if there are any more

makeworld-the-better-one, 2019-02-12:
I'm going to close the issue, since this specific one is gone. The develop branch does need to be tested, but we can share results in chat.

closed 2019-02-12 by makeworld-the-better-one