tomeshnet/prototype-cjdns-pi #318: AdHoc vs. ad-hoc used across doc, consistent preferred.

darkdrgn2k, 2019-03-03:


darkdrgn2k, 2019-03-04:
ad hoc

Seen in https://tools.ietf.org/html/rfc4260
```
   Basic Service Set (BSS): A set of stations controlled by a single
                  coordination function, where the coordination function
                  may be centralized (e.g., in a single AP) or
                  distributed (e.g., for an ad hoc network).  The BSS
                  can be thought of as the coverage area of a single AP.
```

darkdrgn2k, 2019-03-04:
Ad Hoc

makeworld-the-better-one, 2019-03-04:
Ad-hoc

makeworld-the-better-one, 2019-03-04:
adhoc

Shrinks99, 2019-04-07:
> ad hoc
>
> Seen in https://tools.ietf.org/html/rfc4260
>
> ```
>    Basic Service Set (BSS): A set of stations controlled by a single
>                   coordination function, where the coordination [...]

This is also the way [Wikipedia](https://en.wikipedia.org/wiki/Ad_hoc) uses the phrase with "Ad hoc..." being used at the start of a sentence if need be.

makeworld-the-better-one, 2019-04-07:
@darkdrgn2k Can you remove your double votes? You've voted both up and down on some of them. If you have no opinion, just don't vote.

makeworld-the-better-one, 2019-04-07:
[Wiktionary](https://en.wiktionary.org/wiki/ad%2Dhoc) says that ad-hoc is an alternative spelling of ad hoc, which suggests that we should stick with ad hoc. I feel fine resolving this issue with the conclusion of using ad hoc, and a search tool like [ripgrep](https://github.com/BurntSushi/ripgrep) can be used to find all the occurrences of alternate spellings