tomeshnet/prototype-cjdns-pi #319: Add configuration to prototype

darkdrgn2k, 2019-03-04:
Proposed specs for configuration

- Leverage standard Debian config as much as possible
- Create config files instead of hard coding values
  - Maybe Use [confget](https://devel.ringlet.net/textproc/confget/) 🤷‍♂️ 
- Document each config option in grouped module sections for easy extraction later

darkdrgn2k, 2019-03-04:
# confget
- Very low dependency requirements
- Existing Systemd format (INI format)
- Easy to read format
```
[section]
config1=value1
config2=values2
config3=value3

[section3]
config1=value1
config2=value2
config3=value3
```
- Easy format to read value
`confget myfile.conf -s  section3 config3`

##  Issues with it:
- [ ] 👎 Does not have a  `confset` 

- In the past I removed a whole section and just re-wrote it [example](https://github.com/darkdrgn2k/prototype-cjdns-pi/blob/30517f8a0037a3d47a1f9eb3068b5978e0f48461/scripts/mesh-point/mesh.config#L24)

makeworld-the-better-one, 2019-03-04:
Potentially using a file such as `/etc/mesh.conf` or `/etc/mesh.conf.<ext>`.

JSON will have a lot of parsers available, but it's hard for humans to write. I think [HJSON](https://hjson.org/) is a nice solution, it's what Yggdrasil uses as well. With HJSON we can do comments, and write nice clean config files, but it's compatible with regular JSON. There are easy converters, and so this means that any existing JSON editing tools can work once you convert the file. Some will support HJSON already.

Example:

```hjson
{
  // use #, // or /**/ comments,
  // omit quotes for keys
  key: 1
  // omit quotes for strings
  contains: everything on this line
  // omit commas at the end of a line
  cool: {
    foo: 1
    bar: 2
  }
  // allow trailing commas
  list: [
    1,
    2,
  ]
  // and use multiline strings
  realist:
    '''
    My half empty glass,
    I will fill your empty half.
    Now you are half full.
    '''
}
```

darkdrgn2k, 2019-03-04:
I looked hjson at this when trying to edit the yggdrasill config.  And everything was not-deb friendly. meaning i had to install some NPM  or pip or compile. I ended up installing jq and converting it to json.

one thing we have to keep in mind is dependencies. If you keep the very like less people will complain.

we already have python only works on prototype. with deb packaging you may not have python and now to parse a config you will have to install it. 

`perl` (as much as we hate it) is "already there" and wont go anywhere because Debian depends on it way to much. So we can assume its there.  Bash probably is a little more iffy but i think it safe that its there to.

On distros like mesh-orange installing something like python could be a deal breaker.

confget as an example is a small 36kb (source code) c program

makeworld-the-better-one, 2019-03-04:
HJSON doesn't need python, look at [this](https://hjson.org/users-bin.html) easy go binary install. But because it's not easily editable maybe it's not good for a config file either.

darkdrgn2k, 2019-03-05:
> HJSON doesn't need python, look at [this](https://hjson.org/users-bin.html) easy go binary install. But because it's not easily editable maybe it's not good for a config file either.

Looks interesting, Added as possibility.  A little big but may be "fixable" by not statically linking it.

makeworld-the-better-one, 2019-03-05:
Talking about HJSON:
> Does read and write

This isn't technically true. HJSON is more for human writing. If you want to do machine reading and writing, the way to do it would be to convert it into JSON, edit it with one of the many existing editors, and then convert it back. As for reading, you could do the same, but there are some readers that will accept it I think.

makeworld-the-better-one, 2019-03-05:
Now I'm thinking YAML is the best way to go. Very easy to read and write for humans, and with the [yq](https://mikefarah.github.io/yq/) tool, machine read write is very easy too. yq is only 4.69MB and written in go.

darkdrgn2k, 2019-03-08:
>  4.69MB
5 megs is allot when you have a 512 mb ram board and need to use that for the RAM DISK as well.

makeworld-the-better-one, 2019-03-09:
Well maybe we can write a reader/writer ourselves? YAML seems like the best format to me, but maybe it doesn't have enough tools. What are you thinking we should do? It looks like you fixed the `confget` issue, so maybe we can just use INI type files?

darkdrgn2k, 2019-03-24:
I was playing with a bash implementation of "setConf" seems to work pretty good.

https://github.com/tomeshnet/prototype-cjdns-pi/blob/ygg-tunnel-ipv6/scripts/yggdrasil-iptunnel/yggdrasil-setup#L7

makeworld-the-better-one, 2019-03-24:
Nice. If both get and set work nicely, then INI style sounds okay to me

makeworld-the-better-one, 2019-04-21:
A config file has begun taking shape in #394.