tomeshnet/prototype-cjdns-pi #322: Yggdrasill internet peer and ip tunnel conflict

darkdrgn2k, 2019-03-04:
If yggdrasill is using an interent peer, and set to ipclient it will fail

Reason

1. yggdrasil will establish a connection to the internet peer
1. yggdrasill will establish the iptunnel over to the yggdrasill exit node
1. yggdrasill will run all traffic over the new iptunnle to the exit node
1. yggdrasill will try to send all peering trafic over the exit node
1. yggdrasill peering will fail because its tries to send info over the ip tunnel the is depended on the peer  that is now depended on the iptunnel
1. It will stop working

work around - createa a route over your internet connection to he exit peer

`sudo ip route add 1.1.1.1 gw 2.2.2.2`

where 1.1.1.1 is your internet peer and 2.2.2.2 is your internet gateway

makeworld-the-better-one, 2019-04-13:
Has this been resolved? I saw documentation about doing exactly this in a recent PR, #377 maybe.

darkdrgn2k, 2019-07-19:
Aside from note in the docs it is NOT "fixed"

closed 2019-10-27 by darkdrgn2k