tomeshnet/prototype-cjdns-pi #323: Configuration of WIFI

darkdrgn2k, 2019-03-04:
Related to #317 and #319 

Configuration of WiFi adapters is problematic:

**WLANx moves around**
- You cant guarantee (for example) that onboard wifi will always be wlan0 and a use will be wlan1. Sometimes wlan0 will be usb on boot
- two usb devices may note be wlan0 and wlan1

**Current Access Point Workaround**
- Access point is renamed to wlan-ap using the onboard driver. There will always? be only way onboard-driver on the board so it works

**Locking wireless devices by mac address**
- Some devices have mac addresses that change between reboots :( 
- USB devices that are swapped between boards

**problem**

Cant use macs - as they are not persistent enough
Cant use drivers cause you cant have two same type devices do two different things

ISSUE: How do you identify what device is doing what in the config.

simplymathematics, 2019-06-20:
In Ubuntu, there's an install option about making predictable interface names. I wonder if raslbian/Debian have the same. 

It works by numbering the interfaces by Mac address.

darkdrgn2k, 2019-07-21:
predictable is the first thing we disable because it's not that predictable esp with broken hardware that screws up.mac addresses.