tomeshnet/prototype-cjdns-pi #325: Services Dependencies

darkdrgn2k, 2019-03-04:
Break service hacks for before and after into their own services

`After=`/`Before=` can be used not to create Dependencies but still set order

Some places to fix this

- IPFS Swarm
- IPTunnel cjdns & yggdrasill

darkdrgn2k, 2019-03-09:
Before/After dont work son servivcerestarts

look into BIND instead

Drop-in module can also be used 

See example 
https://github.com/tomeshnet/prototype-cjdns-pi/blob/ygg-tunnel-ipv6/scripts/yggdrasil-iptunnel/install#L11-L14

makeworld-the-better-one, 2019-04-27:
Can you elaborate on what you mean here?

darkdrgn2k, 2019-07-27:
question is how to create a seperat deb for (example) cjdns and iptunnel

currently iptunnel modifies cjdns service file but that's invasive 

if you create a separate service and you use after= or before= sets the order of which service starts when

for example
start cjdns
start cjdns iptunnel

however when cjdns is restarted cjdns iptu  do won't restart

using bind could be used instead. that should marry two services.togetber

when cjdns stops cjdns iptunnel should stop 

I been looking I to dropin where you change an existing service by adding a separate file into a folder

this would have the same.effect as what is currently being done without the invasive practices