tomeshnet/prototype-cjdns-pi #326: Module ideas

makeworld-the-better-one, 2019-03-05:
From the HackMD **Modules** section:

- No flashing for updates
- Individual packages
- Pre-compiled where possible
    - Only when the original maintainers do this
    - Otherwise it can't be verified that we aren't doing anything malicious
- Can be installed, uninstalled, updated separately from each other
- Easily accessible package scripts online
- Hosted on the Internet and mesh
- Updates over mesh possible
- Decentralized updates maybe
    - IPFS
    - Packages would be cached by nodes when they update or install stuff
    - Local IPFS address in apt conf
- Likely as `.deb` packages
- Module packaging standards and rules
- Other people adding modules
    - Anyone can make a PR to add to our module repo
    - If it is rejected for not following standards/rules/COC then anyone can make their own repo and add it into their node
        - Like how many Linux OS package repos work

darkdrgn2k, 2019-03-05:
> No flashing for updates
> Individual packages

Debian repository already decided upon as the direction we will go. 

>Can be installed, uninstalled, updated separately from each other

Debian repository. Obviously they will need to satisfy dependencies.

Checked off items have already been decided on based on the direction agreed upon (DEB repository)

> Pre-compiled where possible 
> Only when the original maintainers do this
> Otherwise it can't be verified that we aren't doing anything malicious
> Easily accessible package scripts online

The package build script will be available as open source repository and can be peer reviewed. Other considerations can be made when the repository is created  about how the deb packages actually get built/released.

> Decentralized updates maybe 
> Hosted on the Internet and mesh

This is dependent on the how the repository is created/hosted and what it can do.

> Other people adding modules 

These will be rules set out at when the Debian repository is created. We would have to have some sort of checks/balances to keep the purity of the repository.

However

- Packages can always be created but not necessarily committed to our repo
- Anyone can host their own Repo
- Repo deployment will be part of Mesh Services and replicatable

Shrinks99, 2019-03-08:
> Decentralized updates maybe
> Hosted on the Internet and mesh

Could [apt-p2p](https://manpages.ubuntu.com/manpages/bionic/man8/apt-p2p.8.html) be used for this?  Seems interesting.

makeworld-the-better-one, 2019-03-08:
@Shrinks99 I am against including another p2p system, another long-running service, DHT, etc. I'm thinking with stick with IPFS, since we are already using it.

Shrinks99, 2019-04-07:
[IPFS is focusing on package managers this year](https://blog.ipfs.io/78-ipfs-2019-roadmap/) and [apt IPFS mirrors](https://github.com/ipfs-shipyard/apt-on-ipfs) also exist.

Something to look into after we have everything converted to Debian packages and a centralized (regular) repo working.