tomeshnet/prototype-cjdns-pi #335: No internet after node reboot.

Ikata89, 2019-03-06:
I had an issue with my node connecting to internet after reboot. Had the same issue before and deleting _**/var/run/yggdrasil.sock**_ + restart of the service(_**sudo systemctl restart yggdrasil**_) fix it. 

This file was deleted before that for the same reason, but it came back after the reboot.

darkdrgn2k, 2019-03-06:
> This file was deleted before that for the same reason, but it came back after the reboot.

File is created when yggdrasill starts up. Its the means yggdrasill communicates with yggdrasillctl and such

File should be deleted when Yggdrasil stops. Seems it crashed and caused this file to remain.

darkdrgn2k, 2019-03-06:
Issue is already patched in develop

https://571-115685026-gh.circle-artifacts.com/0/yggdrasil-develop-0.3.3-0054-armhf.deb

neilalexander, 2019-03-06:
This *should* have been addressed in yggdrasil-network/yggdrasil-go#356. Please let us know if it's still a problem in any version of `yggdrasil-develop` after version `0.3.3-0035`.

Ikata89, 2019-03-06:
Temporary solution: 

1. Download Yggdrasil dev by **_wget https://571-115685026-gh.circle-artifacts.com/0/yggdrasil-develop-0.3.3-0054-armhf.deb_**

2. Extract the file **_dpkg -i yggdrasil-develop-0.3.3-0054-armhf.deb_**

3. Go to **_/prototype-cjdns-pi/scripts_** and run **_bash yggdrasil-iptunnel/install_**

4. Edit _yggdrasil-pre-setup_ as there is a small bug in the install -> **_sudo nano /usr/local/sbin/yggdrasil-pre-setup_** and replace line 15 which is  _**if [[ "${PUBLIC_KEY}" =~ ^[0-z]{64}]];**_ with **_if [[ "${PUBLIC_KEY}" =~ ^[0-z]{64} ]];_**. The difference is a SPACE after {64} and before ]]

5. Reboot and if everything was done OK you should be sorted.

darkdrgn2k, 2019-03-06:
solution edit: The error in number  4 is an isolated issue in only a few commits and has been fixed.

darkdrgn2k, 2019-04-18:
Issue seems to be resolved. Closing.

closed 2019-04-18 by darkdrgn2k