tomeshnet/prototype-cjdns-pi #342: Debian 9 missing rc.local

darkdrgn2k, 2019-03-11:
Bring it back

https://www.itechlounge.net/2017/10/linux-how-to-add-rc-local-in-debian-9/

or work around it.

makeworld-the-better-one, 2019-03-12:
Unless there's a supported alternative that isn't just "Create a systemd unit file for every damn thing", I would vote to bring it back.

makeworld-the-better-one, 2019-04-27:
What is being done about this? My current RPi install is on Raspbian 9 and has `/etc/rc.local`, but I guess other pure Debian installs don't?

```
pi@tomesh-d15b:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.8 (stretch)
Release:	9.8
Codename:	stretch
pi@tomesh-d15b:~ $ cat /etc/rc.local
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

service dnsmasq restart
rfkill unblock wifi && service hostapd restart
/usr/local/bin/nodeinfo-update.sh
/usr/local/bin/nodeinfo-update.sh
exit 0
```

darkdrgn2k, 2019-06-16:
Correct

I guess debian added it so i can output your ip address :)

makeworld-the-better-one, 2019-06-19:
So what are we doing for non-raspbian installs? Or is it not necessary in our stack anymore?