tomeshnet/prototype-cjdns-pi #349: Add SSB announce of pub

darkdrgn2k, 2019-03-13:
https://github.com/ssbc/docs/blob/master/scuttlebot/howto-announce-a-pub-server.md

darkdrgn2k, 2019-03-14:
Need more explenation what this actaully does.

Seems to be added to clients when they are added to a pub to announce the existence of the pub.

makeworld-the-better-one, 2019-03-15:
Asked about it on Scuttlebutt [here](https://viewer.scuttlebot.io/%25WdTNUN7dJ01MWgXT%2B868NdsA25SvJt7ZTIhnDWQKqXI%3D.sha256). The hash is `%WdTNUN7dJ01MWgXT+868NdsA25SvJt7ZTIhnDWQKqXI=.sha256`, and I tagged you @darkdrgn2k.

makeworld-the-better-one, 2019-03-16:
As discussed in chat, this is likely not needed for our nodes to do. Clients will do this automatically.

darkdrgn2k, 2019-04-05:
Pubs already announced during each used invite.

No need for more announcements.

closed 2019-04-05 by darkdrgn2k