tomeshnet/prototype-cjdns-pi #351: Links and table in readme break in tomesh.net

dasanchez, 2019-03-14:
The "Build A Node" page of tomesh.net embeds the readme of this repo as follows:
```
{% remote_markdown https://raw.githubusercontent.com/tomeshnet/prototype-cjdns-pi/master/README.md %}
```
- The Frequently Asked Questions and MODULES.md links lead to 404s.
- The Hardware Table is not formatted properly.

darkdrgn2k, 2019-03-14:
404 errors fix attempt in this branch - https://github.com/tomeshnet/tomesh.net/tree/remote-markdown-patch

[Diff](https://github.com/tomeshnet/tomesh.net/compare/remote-markdown-patch?expand=1)

Don't know if it works.  I know there was discussion

dasanchez, 2019-03-15:
Thanks @darkdrgn2k I'll test that

darkdrgn2k, 2019-03-16:
DOH I Edited dasanchez msg sorry . put it back :)

FAQ error was in the code #353 fixed 

Perhaps "build a node" should be written up in a better way then just linking to prototype.
Possibly related to tomeshnet/tomesh.net/issues/215 point 2

dasanchez, 2019-03-26:
@garrying fixed the broken links issue  with [PR 252](https://github.com/tomeshnet/tomesh.net/pull/252).

closed 2019-03-26 by dasanchez

garrying, 2019-03-27:
Fix for the table issue in https://github.com/tomeshnet/prototype-cjdns-pi/commit/33354d3262eddf3b9391f555079a773f5ff1ff17