tomeshnet/prototype-cjdns-pi #364: IPFS Install issue

darkdrgn2k, 2019-03-23:
IPFS wont install in develop
```
go-ipfs/build-log
go-ipfs/install.sh
go-ipfs/ipfs
go-ipfs/LICENSE
go-ipfs/README.md
initializing IPFS node at /home/pi/.ipfs
generating 2048-bit RSA keypair...done
peer identity: QmQoPyXSEHF6VMMGUS4Hi7MrPj4MdvxrJKZ4ssHjYY7Pay
to get started, enter:

	ipfs cat /ipfs/QmS4ustL54uo8FzR9455qaxZwuMiUhyvMcX9Ba8nUH4uVv/readme

Error: failed to set config value: failure to decode config: json: cannot unmarshal string into Go struct field ConnMgr.LowWater of type int (maybe use --json?)
```

darkdrgn2k, 2019-03-23:
```
+ ipfs config Pubsub.Router gossipsub
+ ipfs config --bool Experimental.FilestoreEnabled true
+ ipfs config Swarm.ConnMgr.Type basic
+ ipfs config Swarm.ConnMgr.LowWater 100
Error: failed to set config value: failure to decode config: json: cannot unmarshal string into Go struct field ConnMgr.LowWater of type int (maybe use --json?)
```

makeworld-the-better-one, 2019-03-24:
Oh yeah I had this issue on my local machine, I'll fix it in a new PR. The number changes need the `--json` flag as the CLI suggests.

PR #367 (2019-03-24): IPFS JSON fix
Closes #364
closed 2019-03-24 by darkdrgn2k