tomeshnet/prototype-cjdns-pi #368: Migrate hostname and meshname to config

darkdrgn2k, 2019-04-03:
https://github.com/tomeshnet/prototype-cjdns-pi/blob/ygg-tunnel-ipv6/scripts/hostapd/install#L48
https://github.com/tomeshnet/prototype-cjdns-pi/blob/ygg-tunnel-ipv6/scripts/install2#L409
https://github.com/tomeshnet/prototype-cjdns-pi/blob/ygg-tunnel-ipv6/scripts/cjdns/install#L126
https://github.com/tomeshnet/prototype-cjdns-pi/blob/ygg-tunnel-ipv6/scripts/yggdrasil/install#L65
https://github.com/tomeshnet/prototype-cjdns-pi/blob/ygg-tunnel-ipv6/scripts/install2#L402

makeworld-the-better-one, 2019-04-03:
Are you referring to the config file referenced in #319? Might be better to put those links there instead of having the separate issue, since we still would need to figure that out before we do this.

darkdrgn2k, 2019-04-05:
Yes,

Currently there is a dependency in some modules that require a host name to have a global variable set. So if you manually install (for example) hostapd, it will not set the configure correctly because the hostname was set in a different module

Using a common config file allows us to remove the need for having this dependency and instead rely on a single file.

hostname and mesh name are the two variables im aware of that right now need global variables set.

makeworld-the-better-one, 2019-04-07:
Yeah that makes sense, thanks for outlining these places.

darkdrgn2k, 2019-06-16:
All done

PR #394 (2019-05-12): Add configurations paramaters
Store Mesh name in /etc/mesh.conf not environment variable

closes #368 
closes #383 
closes #382
closed 2019-06-16 by darkdrgn2k