tomeshnet/prototype-cjdns-pi #376: Control file to change hostname

darkdrgn2k, 2019-04-08:
```
oldhostname=$(hostname)
NEWHOSTNAME="test123"

sudo echo "$NEWHOSTNAME" | sudo tee /etc/hostname > /dev/null
sudo sed -i -e "s/$oldhostname/$NEWHOSTNAME/" /etc/hosts
sudo sed -i -e "s/$oldhostname/$NEWHOSTNAME/" /etc/yggdrasil.conf
sudo sed -i -e "s/$oldhostname/$NEWHOSTNAME/" /opt/tomesh/nodeinfo.json
```

makeworld-the-better-one, 2019-04-10:
No CJDNS needed I assume?

makeworld-the-better-one, 2019-04-10:
Maybe we should have a module that aliases the hostname command to do this in `/etc/profile` or something? Do we want to make this a module?

darkdrgn2k, 2019-04-13:
I don't think you want to do hostname. changed I  profile.

to many services depend on it. hostnames shouldn't be changed randomly either.

in a config screnario it should suggest a reboot after being changed.

makeworld-the-better-one, 2019-04-13:
> I don't think you want to do hostname. changed I profile.

I don't know what you are saying here.

> to many services depend on it. hostnames shouldn't be changed randomly either.

Why did you make this script in the first place? Just in case?

darkdrgn2k, 2019-04-14:
> Why did you make this script in the first place? Just in case?

Came as part of my backup/restore script i was toying with. You would need to rename the node when you restore your cjdns keys.

>

>    I don't think you want to do hostname. changed I profile.

I meant you dont want to change the host-name in the profile.  This requires the user to login to trigger this.   Hostnames are set/changed much earlier in the boot process.

But its not a problem if we dont need to keep changing them :)

makeworld-the-better-one, 2019-04-19:
> I meant you dont want to change the host-name in the profile. This requires the user to login to trigger this. Hostnames are set/changed much earlier in the boot process.

Alright then a systemd unit maybe?

But maybe this issue should just be closed, and you can add this as a contrib script.

makeworld-the-better-one, 2019-04-27:
What are you thinking of doing with this file? There is some hostname stuff going on now in #394, and this should be a contrib file if #394 doesn't do enough. Can this issue be closed?