tomeshnet/prototype-cjdns-pi #395: Prepare modules for debian

darkdrgn2k, 2019-04-21:
Following things should be separated in modules to make conversion to deb easier

- Separate parts of the module that are tomesh specific from generic module
- Separate different parts that will be different modules 
- Separate installation files from post install script and pre install script

makeworld-the-better-one, 2019-04-21:
Should this be done before the v0.5 release, or during v0.6 as we make the debs?

darkdrgn2k, 2019-06-16:
started working on proof of concept [here](http://github.com/darkdrgn2k/packages/) with automatic build and deploy to releases and a DEB repo

Working document on deb repo
https://cryptpad.fr/code/#/2/code/view/klu8+dWdreYHvVvvKo-Gd+10Z6Q+wEerdSR9QHIZs7s/