tomeshnet/prototype-cjdns-pi #396: Keeping .ipfs directory then installing

makeworld-the-better-one, 2019-04-23:
```
Saving to: ‘/home/pi/prototype-cjdns-pi/scripts/ipfs/tmp/go-ipfs.tar.gz’

/home/pi/prototype-cjdns-pi/scripts/ipfs/tmp/go 100%[=====================================================================================================>]  16.85M  3.28MB/s    in 37s     

2019-04-23 01:29:25 (469 KB/s) - ‘/home/pi/prototype-cjdns-pi/scripts/ipfs/tmp/go-ipfs.tar.gz’ saved [17668228/17668228]

go-ipfs/install.sh
go-ipfs/ipfs
go-ipfs/LICENSE
go-ipfs/README.md
Error: failed to take lock at /home/pi/.ipfs: permission denied
Error: cannot connect to the api. Is the deamon running? To run as a standalone CLI command remove the api file in `$IPFS_PATH/api`
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs init
Error: failed to take lock at /home/pi/.ipfs: permission denied
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs init -v
Error: unknown option "v"
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs init --verbose
Error: unknown option "verbose"
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs init
Error: failed to take lock at /home/pi/.ipfs: permission denied
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs config Pubsub.Router gossipsub
Error: cannot connect to the api. Is the deamon running? To run as a standalone CLI command remove the api file in `$IPFS_PATH/api`
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ rm ~/.ipfs/api
rm: remove write-protected regular file '/home/pi/.ipfs/api'? y
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs config Pubsub.Router gossipsub
Error: cannot acquire lock: Lock Create of /home/pi/.ipfs/repo.lock failed: open /home/pi/.ipfs/repo.lock: permission denied
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ls ~/.ipfs
blocks  config  datastore  datastore_spec  keystore  repo.lock  version
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ cat ~/.ipfs/repo.lock
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ rm ~/.ipfs/repo.lock
rm: remove write-protected regular empty file '/home/pi/.ipfs/repo.lock'? y
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs config Pubsub.Router gossipsub
Error: open /home/pi/.ipfs/config: permission denied
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ls -la ~/.ipfs
total 36
drwxr-xr-x   5 pi   pi   4096 Apr 23 01:34 .
drwxr-xr-x  11 pi   pi   4096 Apr 23 01:14 ..
drwxr-xr-x 245 pi   pi   4096 Mar 26 03:16 blocks
-rw-rw----   1 root root 5839 Mar 24 01:08 config
drwxr-xr-x   2 pi   pi   4096 Apr 21 21:10 datastore
-rw-------   1 pi   pi    190 Feb 17 01:35 datastore_spec
drwx------   2 pi   pi   4096 Feb 17 01:35 keystore
-rw-r--r--   1 pi   pi      2 Feb 17 01:35 version
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ cat config
cat: config: No such file or directory
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ cat ~/.ipfs/config
cat: /home/pi/.ipfs/config: Permission denied
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs daemon
Initializing daemon...
go-ipfs version: 0.4.20-
Repo version: 7
System version: arm/linux
Golang version: go1.12.4

Error: open /home/pi/.ipfs/config: permission denied
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ sudo chown pi ~/.ipfs/config
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs config Pubsub.Router gossipsub
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs daemon
Initializing daemon...
go-ipfs version: 0.4.20-
Repo version: 7
System version: arm/linux
Golang version: go1.12.4
Swarm listening on /ip4/10.0.0.1/tcp/4001
Swarm listening on /ip4/127.0.0.1/tcp/4001
Swarm listening on /ip4/192.168.50.242/tcp/4001
Swarm listening on /ip6/204:3d1b:9347:2f52:4021:4198:7690:d283/tcp/4001
Swarm listening on /ip6/304:3d1b:9347:2f52::1/tcp/4001
Swarm listening on /ip6/::1/tcp/4001
Swarm listening on /p2p-circuit
Swarm announcing /ip4/10.0.0.1/tcp/4001
Swarm announcing /ip4/108.168.59.253/tcp/4001
Swarm announcing /ip4/127.0.0.1/tcp/4001
Swarm announcing /ip4/192.168.50.242/tcp/4001
Swarm announcing /ip6/204:3d1b:9347:2f52:4021:4198:7690:d283/tcp/4001
Swarm announcing /ip6/304:3d1b:9347:2f52::1/tcp/4001
Swarm announcing /ip6/::1/tcp/4001
API server listening on /ip4/127.0.0.1/tcp/5001
WebUI: http://127.0.0.1:5001/webui
Gateway (readonly) server listening on /ip4/127.0.0.1/tcp/8080
Daemon is ready
^C
Received interrupt signal, shutting down...
(Hit ctrl-c again to force-shutdown the daemon.)
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs config Pubsub.Router gossipsub
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ls ~/.ipfs
blocks  config  datastore  datastore_spec  keystore  version
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ touch ~/.ipfs/repo.lock
pi@tomesh-2c52:~/prototype-cjdns-pi/scripts $ ipfs config Pubsub.Router gossipsub
```

makeworld-the-better-one, 2019-04-23:
This might not matter, since in our system we usually reflash, but it's still important for deb stuff.

darkdrgn2k, 2019-05-12:
Not sure why your getting permissions denied.

But probably because  IPFS  runs as ROOT currentyl even though the .ipfs directory is in /home/pi/.ipfs

makeworld-the-better-one, 2019-05-18:
Fixed in develop

PR #402 (2019-05-13): Prevent re-initalize of ipfs datastore if it exists
Closes #396
closed 2019-05-18 by makeworld-the-better-one